guysoft/FullPageOS #102: WebGL support?

peteruithoven, 2017-02-03:
I'm trying to get WebGL working. Visiting http://get.webgl.org/ showed that it wasn't enabled yet. 

I followed https://www.scirra.com/blog/ashley/24/a-better-way-to-get-webgl-on-the-raspberry-pi-2, but discovered I also had to install `xcompmgr`, but afterwards http://get.webgl.org/ didn't yet show the cube.
 
I then continued with https://www.scirra.com/blog/ashley/23/how-to-get-webgl-on-the-raspberry-pi-2, but getting to the chrome://flags page seems tricky. After I inserted this in the fullpageos.txt file, after reboot, I got the regular start screen of Chromium. I also noticed that I couldn't use my wireless keyboard, which worked fine on Raspbian, to type in another address. I could use it's mouse functionality. 

This is where I'm stuck. Is there another way to enable the `Override software rendering list` flag? 
Could I use VNC to get to Chromium's flags page? I only get to the commandline using vnc.

guysoft, 2017-02-03:
What version of FullPageOS? Did you try version 0.7.0RC1?

peteruithoven, 2017-02-03:
Apologies, I'm trying what seemed to me to be the latest stable (http://docstech.net/FullPageOS/2016-07-20_2016-05-27-fullpageos-jessie-lite-0.7.0.zip)
Should I try 0.7.0RC1?

guysoft, 2017-02-03:
Use this one. Its the latest build, its not stable but should have webgl.
If you can update success on the issue it would help me undressed if I
should release it:
https://github.com/guysoft/FullPageOS/issues/92

peteruithoven, 2017-02-03:
Looks awesome! I will try and get back to you. 

(I did get vnc working, using Vinagre, I forgot to switch the protocol from ssh to vnc. But also there I couldn't type in an address.)

peteruithoven, 2017-02-03:
[image]

Nice. That 0.7.0 RC1 works out of the box. I didn't even need to enable open GL. It also responded to keyboard input.

guysoft, 2017-02-03:
Good, would be helpful if you can give thumbs up or down at #92

closed 2017-02-03 by guysoft

peteruithoven, 2017-02-03:
Done, but I'm afraid I don't have enough reputation on raspberrypi.stackexchange.com to give upvotes that are publicly shown.