guysoft/FullPageOS #104: Rectangle Font instead of Gujarati/Hindi text Font

G0p14n1, 2017-02-14:
I have loaded a URL( http://viewer.headsup.news/?channelId=58346107866769ee6aa26994) of a full screen RSS Page Displaying text in English and Gujarati language text  perfectly displayed on chromium browser using RPI Raspbian OS Jessie.

But when using FullPageOS (2016-05-10-fullpageos-jessie-lite-0.6.0) I get Rectangle Box displayed instead of the Gujarati/Hindi language text , English Text is displayed Perfectly .Is there any way to get this text font showing correctly
[image]
 ?

Thanks

guysoft, 2017-02-15:
Is there a hindi language package for Raspbian?

G0p14n1, 2017-02-19:
I have no idea but when I use the chromium browser on raspbian pixel latest OS  it is displayed    correctly  but when booting the url with full pageOS it's displayed  as rectangle  boxes. Is there a language  font/text option/setting that I could change for me to view the text correctly  or will this bug be corrected  in the next l future update ? 

Thank you for  Fullpage OS.

guysoft, 2017-02-19:
What about installing 
```
apt-get install fonts-deva fonts-deva-extra fonts-gujr fonts-gujr-extra
```
That might help

G0p14n1, 2017-02-19:
Hi could you kindly please instruct me on how to install this  "sudo apt-get install fonts-deva fonts-deva-extra fonts-gujr fonts-gujr-extra" as i am a N00b in this area. 

Thanks

guysoft, 2017-02-20:
I suggest you read:
https://help.ubuntu.com/12.04/serverguide/apt-get.html

G0p14n1, 2017-02-22:
Hi none of this makes any  sence to me,How do i get to the  "command-line tool" if this OS boots straight in to the Browser when booted up ?,Please help.

guysoft, 2017-02-22:
Assuming you are using windows:
https://www.raspberrypi.org/documentation/remote-access/ssh/windows.md

I'm sorry, I can't teach you how to use linux, not because I don't want to, but because its like teaching a how to use Arduino form scratch, or how to build things, it takes skill. Its something you can learn the basics in just a few weekends. But its outside the scope of this issue, and its not a FullPageOS issue. Moreover I have a company to run, I am doing this as an volunteer, no one is paying me for maintain FullPageOS and this issue tracker.

Adding language fonts is, and this is why this issue is still open.

G0p14n1, 2017-02-23:
its no problem I can understand, Anyway I will try this out and will let you know what happens.

Once again thank you for Full page OS you a doing an Excellent job Keep up the good work. :)

guysoft, 2017-03-11:
You need to edit your HTML/CSS for that. And its not an FullPageOS issue.
Another method would be change the DPI of lightdm. [Here are instructions on that](http://askubuntu.com/questions/72486/can-i-pass-a-dpi-flag-to-lightdm-to-set-the-proper-dpi-for-x)

guysoft, 2018-07-25:
Closing if no reply