guysoft/FullPageOS #106: Full Inode

r0n22, 2017-02-20:
I have multiple FullpageOS raspberry Pi 3 running. I am currently running into chromium filling all inodes in / directory with cache files.  All stored in ~/.cache/chromium 

This shows it self as a black screen when raspberry pi is turned on as chromium  is in an infinite boot loop.

Has anyone else seen this and know any possible fixes?

guysoft, 2017-02-20:
How large is your SD card, and did it resize it after boot?
What version of FullPageOS are you using?

r0n22, 2017-02-21:
8 gb and Yes I did.  Using version 0.6.0

guysoft, 2017-02-21:
1. You can limit chromium cache size in two ways ([source](http://stackoverflow.com/questions/40507732/limit-chromium-cache-size-in-linux)):
    a.  Add the variable `` `` , in the file ``/home/pi/scripts/start_chromium_browser`` , for example: ``chromium-browser --disk-cache-size=500000000`` for 500MB 
    b. You can check the maximum value in ``chrome://net-internals/#httpCache``
2. Can you try using 0.7.0 RC1 and see if it happens there? There newer version of chromium might handle

r0n22, 2017-02-21:
1) It seems that it was not filling the disk it was filling the number of files allowed on the disk.  I still had about 70% free on the disk but could not create any more files.  So it was not the size but the number.

2) I will test out the next version but this is over about 3 months of 24/7 usage for this to occur so I will close this as I don't want clutter while I test.

closed 2017-02-21 by r0n22

guysoft, 2017-02-21:
Did you mean to close the issue? Doesn't sound like its closed.