guysoft/FullPageOS #107: How to recover from browser crash?

cesarzapatawfm, 2017-02-27:
I leave my RPI'S running FullPageOS 0.6.0 for days but sometimes Chromium just crashes ( my guess) and it just shows a blank page. The wifi works, the site works but chromium just shows a blank page.  To fix it I just press F5 and that seems to fix it. 

Is there a way to detect the possible crash and recover? a script that runs every 5 minutes and restart chromium if a crash is detected?

tailorvj, 2017-02-28:
This is a good point. I'd like to solve this type of issues by changing the way FullPageOS loads web pages into Chromium. I can't do this myself at the moment though and I'm going to need a JavaScript programmer that can do WebSockets programming to assist me. If anyone is interested in putting a few hours into this please contact me.

SManAT, 2017-05-30:
I've tried ist this was: Search in Chroium Preferences for Errorcodes and Replace it.
```
#Chromium has crashed?
sed -i 's/"exited_cleanly":false/"exited_cleanly":true/' $chromiumPATH/Default/Preferences
sed -i 's/"exit_type": "Crashed"/"exit_type": "None"/' $chromiumPATH/Default/Preferences
```
That worked for me.
But the "I have crashed" Bubble was schowing anyway.
I think you can use this every 5min if you want to.

I came up with this Solution 
```
#overwrite Preferences
cp cleanPreferences $chromiumPATH/Default/Preferences
```
Each time i overwrote the Default Preferences from Chromium.

esaruoho, 2017-08-11:
I've tried to solve this chromium browser info-fetch crashing by creating a crontab on FullPageOS but I can't seem to figure out how to do it. I've set up multiple cron jobs (to reboot at midnight, to reboot a minute from now) but nothing seems to happen.

guysoft, 2017-09-26:
Hey, just a heads up that latest nightly 0.8.0 seems to be much more stable, with less crashes:
http://unofficialpi.org/Distros/FullPageOS/nightly/2017-09-18_2017-09-07-fullpageos-stretch-lite-0.8.zip
http://unofficialpi.org/Distros/FullPageOS/nightly/2017-09-18_2017-09-07-fullpageos-stretch-lite-0.8.zip.md5