guysoft/FullPageOS #108: One pi with one static IP, but getting 2 IPs on network.

FloppyKing, 2017-02-28:
I have my FullPageOS installation setup with the below entry in fullpageos-network.txt in the boot folder.
```
auto eth0
iface eth0 inet static
   address 10.28.0.193
   netmask 255.255.0.0
   gateway 10.28.0.250
   dns-nameservers 10.28.0.6 10.29.0.99 10.3.0.100
```
when i do an ifconfig -a it correctly reports the IP address as 10.28.0.193 on eth0, but when i use a tool like Angry IP Scanner it's showing up as 10.28.0.193 AND 10.28.0.184.  It successfully connects via SSH to both IPs, but it should be only using the .193 that is assigned.

Googling brought up these similar issues linked below which suggest that it is a bug in bug in dhcpcd5 and raspberrypi-net-mods:
http://raspberrypi.stackexchange.com/questions/31526/raspberry-pi-has-two-ip-addresses
https://www.raspberrypi.org/forums/viewtopic.php?f=66&t=111887

Is this something that can be fixed?

FloppyKing, 2017-02-28:
i should mention that I'm using the nightly build
`2016-12-27_2016-11-25-fullpageos-jessie-lite-0.7.0_RC1.zip`

FloppyKing, 2017-02-28:
For now I'm just going to have our DHCP admin assign static IPs to the Pis mac address to match the static ip i'm trying to set since i figure the extra IP is coming from the dhcp even though it shouldn't be.  That way i should only get the one i want.  If i do this, should i remove my entry in fullpageos-network.txt so that it just uses the DHCP?

guysoft, 2017-02-28:
yes. Its plain DHCP and probably setting it via the router is a better
option. Otherwise the router might allocate the address you set in the
static setting.

FloppyKing, 2017-02-28:
so as it stands, it's best to just not use fullpageos-network.txt to set a static IP since it will get you two IPs on your network?

bhartvigsen, 2017-07-31:
Also seeing the same issue here.

guysoft, 2017-07-31:
See instructions here how to set up static IP, they are newer: https://github.com/guysoft/OctoPi/issues/219

bhartvigsen, 2017-07-31:
Thanks for linking that. I ended up finding that I could just 'apt-get remove dhcpcd5' and the unwanted IP went away.

guysoft, 2017-09-26:
Closing since this is fixed.

closed 2017-09-26 by guysoft