guysoft/FullPageOS #110: Custom dashboard

pejko, 2017-03-03:
Hi, thank you for this great distro! I was doing something very simillar - just with raspbian and i was able to run chromium fullscreen by editing the autostart file in home/pi/.config/lxsession/LXDE-pi
..i have my own dashboard (node js ) that i was autostarting by editing the rc.lokal file in etc and just simple adding this lines  
cd /home/pi/myDashboard
node index.js

my question is how to implement my solution to your destro ? 
my dashboard runs on localhost:8000 but and i have my "tabs" on link like : http://localhost:8000/17022810370001wlr1/ 
so what i did was insert my link in to fullpageos.txt 
and run my dashboard with rc.lokal file 
the result is that 3 of 4 times my raspberry loads visually only into  localhost:8000,  but i have a div that shows me the current location ( js function window.location;) and it says that my location is http://localhost:8000/17022810370001wlr1/ 
in 1 of 4 times its loads correctly  
what i think, it might by the problem is the "run_onepageos" script, as i guess here i should disable your dashbord and run mine... i tried it diferent ways but most of the times just ended in debian login screen :/
any ideas how to disable FullPageOsDashboard and run mine ? 
And last question, how can i enable "developer tools" in chromium so i can see my console output and debug my dashboard ?
Sorry for my terrible english and thank you for your time 
Pejko.

guysoft, 2017-03-11:
You can install your dashboard and its dependen.ces, then point `` /boot/fullpageos.txt`` to the url ``http://localhost:8000`` (which you mentioned)

If you want to make your own build, you can use [FullPageOS's variant system](https://github.com/guysoft/FullPageOS#building-fullpageos-variants) to create your own custom distro.

FloppyKing, 2017-03-13:
I'd like to know the answer to his "last question".  How can we enable "developer tools" in chromium?  I need to be able to view the locally created ssl certificate so i can export it and add it to the trusted authorities, but when i try to go to the developer tools it just does nothing as if i hadn't even pressed the button.

pejko, 2017-03-13:
Hi thanks for answer :) @FloppyKing  i managed to enable developer tools by editint the FullPageOS/src/filesystem/home/pi/scripts/start_chromium_browser script ... 
just remove the --kiosk option and chromium will lauch in normal mode where u can use dev tools

FloppyKing, 2017-03-13:
Ok, thank you pejko you saved me a lot of work from having to switch distributions.

closed 2017-03-14 by pejko

pejko, 2017-03-14:
no problem, im happy that i could be once in my life useful :D  @FloppyKing   
@guysoft  i did something simillar but my ulr just not wont pass even after refresh , and since im using a framework (Total.js)its hard to tell where is the problem, but i have the console now so i can debug it myself and close this :)