guysoft/FullPageOS #111: Slideshow is not playing

PCPROFFSEN, 2017-03-08:
Using a a page with a slide show as webpage.
On a normal computer it  runs fine,
In FullPAge OS it runs but stops after a while if I use tabs.
If I set the page to full screen, it doesnt run at all, I have to manually change slide.
Why doesnt it run as on a normal computer, can this be fixed ?

guysoft, 2017-03-08:
Can you provide a url? I don't understand how you are running a slide show on a webpage, there are several ways to do this.

remi-grumeau, 2017-03-10:
This really sounds more like an issue in your slideshow code than in FPOS.
Note that, for performance sake, Chrome-ish browsers usually puts
Javascript code like setInterval on hold in non-active tabs. Also, they can
be issues with anchors in kiosk mode.
So both can be single point of failure.

I'd rather investigate those rather than FPOS...

Remi

esaruoho, 2017-08-11:
@PCPROFFSEN  are you placing multiple URLs to fullpageos.txt or to fullpagedashboard.txt ?

guysoft, 2017-08-11:
Closing if no reply

closed 2017-09-26 by guysoft