guysoft/FullPageOS #116: Looking for description how to secure FullPageOS

reinhrst, 2017-03-26:
If I am to deploy a FullPageOS in a semi-public network, I would like to secure it a bit:

- Change pi user password (know how to do this)
- Allow private-public-key SSH logins only (know how to do this)
- Install firewall (know how to do this)
- Disable x11vnc (Need help with this)
- Disable apache (Need help with this; why is Apache running anyways?)
- Anything else I missed?

I'd be happy to write a short how-to on all of these, if I can get some pointers on the last 3 items. I believe such a document should be linked from the readme, since the internet already has too many devices that have been taken over by bad guys.

guysoft, 2017-03-26:
1. x11vnc: uncomment the line ``x11vnc -many -rfbauth /home/pi/.vnc/passwd &`` in ``/home/pi/scripts/run_onepageos``. Or you can do ``sudo apt-get remove x11vnc``, should work too.
2. You can just ``sudo apt-get remove apache2``
3. If you put a firewall you can block all the ports and save time by installing stuff. I think Thats all.

guysoft, 2018-02-16:
Closing if no reply

closed 2018-07-25 by guysoft