guysoft/FullPageOS #117: Raspberry Pi Zero W

chrisvdb, 2017-03-30:
There are a number of Github issues that imply that the Raspberry Pi Zero is (currently) not supported. I have however not found any statement regarding the new Raspberry Pi Zero W. Is this one supported by FullPageOS?

guysoft, 2017-03-30:
Well, I am not entirely sure, and since I have no hardware to test I really can't tell.
However, I can build a new nightly build (which are done on demand). And you can test for yourself and report what the status is.

Technically from what I can read the Raspberry Pi Zero W should not be able to run Armv7 instructions, but since then we moved to use chromium build supplied by the rip-foundation, and they might have fixed something, so its worth testing.

[I've started a build](http://gnethomelinux.com:8080/job/FullPageOS/52/console), when its ready an uploaded I'll link to it here so you can test for yourself.

chrisvdb, 2017-03-30:
Excellent, I will test tomorrow or coming weekend and report back.

chrisvdb, 2017-03-31:
Looks like the build has finished. The link at the bottom does not work, though.

guysoft, 2017-03-31:
Here you go:
https://github.com/guysoft/FullPageOS/issues/118

chrisvdb, 2017-04-03:
I have tried the image without any configuration:

- booted fine
- flashed a square with color gradients for a second
- black screen

What can I do to help debug? Would it help if I sponsor a Raspberry Pi Zero W device?

chrisvdb, 2017-04-05:
Quick update: I have tried the latest [raspian build](https://downloads.raspberrypi.org/raspbian_latest) on my Raspberry Pi Zero W. Everything seems to be functional, including Chromium.

chrisvdb, 2017-04-05:
I have also tested RC1: the screen remained entirely black throughout, not even showing any boot messages in console mode. So, from that perspective RC2 is an improvement over RC1.

thundergreen, 2017-05-07:
Is it possible to have a raspberry Pi1 image?

guysoft, 2017-05-07:
@thundergreen 1. you are off topic
2. No, see https://github.com/guysoft/FullPageOS/issues/43

chrisvdb, 2017-05-16:
Hi @guysoft, is there anything more I can do at this time? For example test a newer build or provide debugging information on the latest build. In the latter case, which information would be helpful?

guysoft, 2017-05-16:
You can try taking RC2 and FIGURE OUT why lightdm does not start there. There is development work to do which I am not reaching. Its beyond just testing.

chrisvdb, 2017-06-09:
I'm planning to do some testing/debugging this weekend. Would you be able to create a new build based on the latest Raspian version?

guysoft, 2017-06-10:
Building, tough RC2 should also work for testing. Little differences. anyway will post a nightly when ready

chrisvdb, 2017-06-12:
@guysoft could you please post the nightly?

chrisvdb, 2017-06-27:
I tried RC3 this morning... same or similar result: blank screen when the graphical environment should be loaded. I'm traveling the next few weeks, so might not be able to test more.

chrisvdb, 2017-06-27:
I have uploaded the logs [here](https://filebin.ca/3RN62gngzBux/RaspberryPiZeroWLogs.zip).

guysoft, 2017-06-27:
In that case the RaspberryPi Zero is ARMc6 and is not supported

closed 2017-06-27 by guysoft

chrisvdb, 2017-06-27:
Hi guysoft, I'm a bit confused... Chromium on the latest Raspbian works perfectly on the Raspberry Zero W, so why could FullPageOS not work as well?

mecatulo, 2023-10-15:
touching this 6 years later, I have a Pi Zero W and I'm getting an error stating that Chromium isn't supported on the RPi Zero W.

is there a way to use a different browser?

guysoft, 2023-10-25:
Zero W is still not supported.
You can try building FullPageOS with firerfox. 
Its a project asked about here and no one picked up the task:
https://github.com/guysoft/FullPageOS/issues/340