guysoft/FullPageOS #118: FullPageOS 0.7.0 RC2

guysoft, 2017-03-31:
FullPageOS 0.7 second release candidate.

The main change from RC1 #92 is the new base image, and the hopes the GPU stuff might be better here.

Since no one has commented on RC1, I can't tell how good it is, only determine from the github issues people keep reporting. 

If no one says this image is a problematic, this time I'll just release it, I am not going to wait three months this time.

iocon, 2017-04-02:
Just started looking at this project this week. 0.6.0 seems to work OK but the slow video is an issue for my application. I tried 0.7.0RC1 and I have the same crash with mp4 that others are reporting.
This version after it boots goes to a black screen with a solid cursor in the top left and appears to lockup. Tried on a Pi2 and a Pi3 and remade the card just in case.

mansj, 2017-04-02:
Can't get it to work, similarly to @iocon I get it booting into a black screen and that's it. 

Pi3 here.

guysoft, 2017-04-02:
Ok, sounds like there is a probelm I'll have to fix

jsevakis, 2017-04-04:
Yeah, same problem as the others -- booting to black screen. Pi 3 model B. Had to revert to RC1.

Looking forward to an update that fixes h.264 playback tho!

guysoft, 2017-04-04:
@jsevakis h.264 playback beyond me (non-mp4 playback works AFAIK), reported it to Rpi devels in their forums, there was a development around February, if you want to not wait (because I an not sure when I'll have time), you can try upgrading RC1 and see if it solves, it [see talks in this link](https://www.raspberrypi.org/forums/viewtopic.php?t=168560&f=63&sid=c567a5a0b9adbd53d6656df659fbf3fd#p1087990)

jsevakis, 2017-04-04:
It appears that upgrading all the packages by just running apt-get dist-upgrade also causes the black screen bootup error from RC2! 

I was able to update JUST Chromium with `sudo apt-get install --only-upgrade chromium-browser rpi-chromium-mods`
Unfortunately YouTube playback still gets the same error. Was there another package I should try to update?

mark-veenstra, 2017-05-16:
Hi. Just tried `0.7.0 RC2`. Attached a HDMI monitor, but it displays a black screen.
Having a look at `/var/log/syslog` in reported:
```
May 16 09:05:37 fullpageos systemd[1]: Stopping Light Display Manager...
May 16 09:05:37 fullpageos systemd[1]: Starting Light Display Manager...
May 16 09:05:37 fullpageos lightdm[722]: ** (lightdm:722): WARNING **: Error getting user list from org.freedesktop.Accounts: GDBus.Error:org.freedesktop.DBus.Er
ror.ServiceUnknown: The name org.freedesktop.Accounts was not provided by any .service files
May 16 09:05:37 fullpageos lightdm[722]: ** (lightdm:722): WARNING **: Could not enumerate user data directory /var/lib/lightdm/data: No such file or directory
May 16 09:05:37 fullpageos systemd[1]: Started Light Display Manager.
May 16 09:05:38 fullpageos systemd[1]: lightdm.service: main process exited, code=exited, status=1/FAILURE
May 16 09:05:38 fullpageos systemd[1]: Unit lightdm.service entered failed state.
May 16 09:05:38 fullpageos dhcpcd[617]: wlan0: no IPv6 Routers available
May 16 09:05:38 fullpageos systemd[1]: lightdm.service holdoff time over, scheduling restart.
May 16 09:05:38 fullpageos systemd[1]: Stopping Light Display Manager...
May 16 09:05:38 fullpageos systemd[1]: Starting Light Display Manager...
May 16 09:05:38 fullpageos lightdm[736]: ** (lightdm:736): WARNING **: Error getting user list from org.freedesktop.Accounts: GDBus.Error:org.freedesktop.DBus.Er
ror.ServiceUnknown: The name org.freedesktop.Accounts was not provided by any .service files
May 16 09:05:38 fullpageos lightdm[736]: ** (lightdm:736): WARNING **: Could not enumerate user data directory /var/lib/lightdm/data: No such file or directory
May 16 09:05:38 fullpageos systemd[1]: Started Light Display Manager.
May 16 09:05:38 fullpageos systemd[1]: lightdm.service: main process exited, code=exited, status=1/FAILURE
May 16 09:05:38 fullpageos systemd[1]: Unit lightdm.service entered failed state.
May 16 09:05:39 fullpageos systemd[1]: lightdm.service holdoff time over, scheduling restart.
May 16 09:05:39 fullpageos systemd[1]: Stopping Light Display Manager...
May 16 09:05:39 fullpageos systemd[1]: Starting Light Display Manager...
May 16 09:05:39 fullpageos systemd[1]: lightdm.service start request repeated too quickly, refusing to start.
May 16 09:05:39 fullpageos systemd[1]: Failed to start Light Display Manager.
May 16 09:05:39 fullpageos systemd[1]: Unit lightdm.service entered failed state.
```
Also checking the `lightdm` service manually (and starting it) keeps failing:
```
pi@fullpageos:~ $ sudo /etc/init.d/lightdm status
● lightdm.service - Light Display Manager
   Loaded: loaded (/lib/systemd/system/lightdm.service; enabled)
   Active: failed (Result: start-limit) since Tue 2017-05-16 09:05:39 UTC; 10min ago
     Docs: man:lightdm(1)
  Process: 736 ExecStart=/usr/sbin/lightdm (code=exited, status=1/FAILURE)
  Process: 731 ExecStartPre=/bin/sh -c [ "$(cat /etc/X11/default-display-manager 2>/dev/null)" = "/usr/sbin/lightdm" ] (code=exited, status=0/SUCCESS)
 Main PID: 736 (code=exited, status=1/FAILURE)

May 16 09:05:38 fullpageos systemd[1]: lightdm.service: main process exited, code=exited, status=1/FAILURE
May 16 09:05:38 fullpageos systemd[1]: Unit lightdm.service entered failed state.
May 16 09:05:39 fullpageos systemd[1]: lightdm.service holdoff time over, scheduling restart.
May 16 09:05:39 fullpageos systemd[1]: Stopping Light Display Manager...
May 16 09:05:39 fullpageos systemd[1]: Starting Light Display Manager...
May 16 09:05:39 fullpageos systemd[1]: lightdm.service start request repeated too quickly, refusing to start.
May 16 09:05:39 fullpageos systemd[1]: Failed to start Light Display Manager.
May 16 09:05:39 fullpageos systemd[1]: Unit lightdm.service entered failed state.
```
Hope this helps, even necessary I can provide more tests/results back. I tried the `Zero W` and `3B` both have same errors and result. I used this build: `2017-03-30_2017-03-02-fullpageos-jessie-lite-0.7.0_RC2.zip`

guysoft, 2017-05-16:
Please use RC1, RC2 is broken. see #92

closed 2017-05-16 by guysoft

guysoft, 2017-06-25:
Use RC3 now #134