guysoft/FullPageOS #119: Build fails with

monotek, 2017-04-04:
#### What were you doing?

./build

#### What did you expect to happen?

Building image

#### What happened instead?

Aborted with error

#### Was there an error message displayed? What did it say?

```
sudo: unable to resolve host laptop
spawn x11vnc -storepasswd
The system has no more ptys.  Ask your system administrator to create more.
    while executing
"spawn x11vnc -storepasswd"
    (file "/home/pi/scripts/setX11vncPass" line 3)
```

#### Version of FullPageOS?

current devel branch

guysoft, 2017-04-04:
Hey,
If you are getting pty problems you should use the Vagrant build method, [see this section in the README](https://github.com/guysoft/FullPageOS#building-using-vagrant).

monotek, 2017-04-04:
OK, ill will give it a try.

guysoft, 2017-07-03:
Any updates? will close if there is no activity.

closed 2017-08-11 by guysoft