guysoft/FullPageOS #120: Not Working with PI 3.5 inch Touchscreen

conrad9900, 2017-04-04:
#### What were you doing?

Trying to get the fullpage to work on the 3.5 inch screen. The LCD-show-161112.tar.gz driver was loaded onto a working fullpage os build from the site here: http://www.waveshare.com/wiki/3.5inch_RPi_LCD_(A)

#### What did you expect to happen?

The screen should try and show the chromium Browser

#### What happened instead?

Once rebooted i get to a console login for pi@fullpageos, no browser boots

#### Was there an error message displayed? What did it say?

Only a Login

#### Version of FullPageOS?

The newest

#### Screenshot(s) showing the problem:

[If applicable. Always include if unsure or reporting UI issues.]

ghorbani-m, 2017-05-09:
i have same problem with lcd touchscreen!

Adoos, 2017-09-18:
Also experience this, I'm thinking its not setup for touchscreen at the moment?

GamerXy1, 2018-05-04:
I used the 10.1 Inch Touch Display from Waveshare. Without touch Driver, the Display works perfectly fine, displaying the Webpage as intended. After I installed the Driver, i got the Login prompt too...
Anyone knows a solution for this?

Mad-Chemist, 2018-08-01:
Seeing a black screen after the color spectrum boot screen

jamesfed, 2018-08-03:
A me too here with the Official 7" Touchscreen on a 3B+ running the various update routines and not getting anywhere.
Get the colour test screen (the initial setup in console displays as well) but after that just a blank display.

regner, 2018-09-19:
Was anyone able to fix this?

regner, 2018-09-19:
Seems related to #247

guysoft, 2018-09-30:
Not sure about that, this is a GPIO screen and #247 is the official touch screen display which uses the ribbon cable, not the same interface, not the same drivers.

woiczi, 2019-04-07:
Any update or solutions for this issue? I´m trying it w/ an GPIO 2.8" Display (WZDPI). All drivers were successfully installed, as I did it on official RPI jessie. Still the output is pushed to the HDMI! It worked fine on official RPI jessie.
Any Idea?!

Thnx.

tailorvj, 2019-04-07:
Hi @woiczi , please connect to the terminal via SSH and run the following commands:
sudo apt-get update
sudo apt-get upgrade
sudo apt-get dist-upgrade
sudo apt-get install raspberrypi-ui-mods
sudo apt-get install raspberrypi-net-mods

You are also welcome to test out the experimental release we are working on at the moment, which will allow you to control your screen network from the cloud. 
Sign up here for download instructions and updates: http://eepurl.com/ghHv4n

Kind regards,
Tailor VJ
FullPageOS product manager

woiczi, 2019-04-08:
Thanks for your quick respond. I did try it. So far, the screen flashes blue, than black... thats it. I did try a few reboots, but no changes. If i connect HDMI, still the picture are displayed on there!
Unfortunately I am in china right now and my connection to international servers are very limited, slowed down or blocked at all. even a VPN does not help. max speed is mostly 186 kb/s so everything I do will take hours or even days. So if you think I can fix this w/o an new installation, I realy would appreciate!  
Thank you for inviting me. I will comply with this when I return to Germany.

tailorvj, 2019-04-08:
Wow, unbelievable how hard it can be to get a proper internet connection in some places. 

Please try using one of the following solutions and let me know how it works. 

[rpi-fbcp](https://github.com/tasanakorn/rpi-fbcp)
[raspi2fb](https://github.com/AndrewFromMelbourne/raspi2fb)

I would also examine the config.txt on the microsd, to make sure HDMI isn't being forced or anything like that. 

Another option you can read about [here](https://bhavyanshu.me/tutorials/force-raspberry-pi-output-to-composite-video-instead-of-hdmi/03/03/2014/) (if it's still relevant), is to use the keyboard to select the display upon device boot. 

Kind regards,
Tailor
FullPageOS product manager

xbeaudouin, 2020-01-02:
Ok find a way to make this working... On /boot/config.txt, just remove or comment 
`
dtoverlay=vc4-fkms-v3d
`
This is perfectly working for me...

Maybe this can be added into the wiki or whatever.
Regards

guysoft, 2020-01-09:
Added to wiki:
https://github.com/guysoft/FullPageOS/wiki

closed 2020-01-09 by guysoft