guysoft/FullPageOS #122: can't enable power saving

ardeus-ua, 2017-04-09:
#### What were you doing?
Tried to install acpid with typical scripts

#### What did you expect to happen?
Expected to get powersaving on 7 inch Rpi Foundation LCD (capacitive touch) connected to Rpi3
#### What happened instead?
power saving is not switching off the LCD in some amount of minutes
#### Was there an error message displayed? What did it say?
N/A
#### Version of FullPageOS?
0.6
[Can be found in /etc/octopi_version ALWAYS INCLUDE.]

#### Screenshot(s) showing the problem:
Not applicable
[If applicable. Always include if unsure or reporting UI issues.]

guysoft, 2017-04-10:
Unfortunately I have no hardware to test this :-/

guysoft, 2018-02-16:
Closing if no reply

closed 2020-02-07 by guysoft