guysoft/FullPageOS #123: multiple FullPageOs in one network. sometimes black screen.

jandeman023, 2017-04-19:
I have two raspberry pi 3 newest version in my network with the same website loading and with the latest version of FullPageOs on it.

In the morning, when the shop opens both of the raspberry pi's start at the same time because of the light switch.

But there is a chance of around 1 out of 6 that one of the two is giving a back screen with a very small white box in the right down border.

The screens and raspberry pi's are build in a wall so it is not very easy to reset the raspbery pi when is gives a black screen. 

The page that the raspberry pi is loading: http://moederdagbon.nl/slider/

Can someone help me?

guysoft, 2017-04-19:
Might be the speed of boot or something, I am not really sure. Can you send a picture?

jandeman023, 2017-04-19:
Thanks for your fast reply. 
do you mean a picture of the black screen?

jandeman023, 2017-04-20:
Here is a picure of the back screen. The white dot in the right down corner. I think it is the cursor.

[image]

Co2p, 2017-04-23:
I have the same issue on a B+

guysoft, 2017-06-30:
@Co2p Raspberrypi 1 B+ is not supported only 2 and up

guysoft, 2017-08-11:
Closing if no answer

Co2p, 2017-08-14:
@guysoft
excuse my slow answer.
I am not sure if it was a pi 1 or 2 that was on.
Should I expect a crashlog on the pi 1 B+, or does it fail silently?
The symptoms I got look exactly like the screenshot.

guysoft, 2017-08-14:
@Co2p Raspberrypi 1 B+ is not supported, I don't know how it fails, because I don't test for it. It should get stuck at some point.

Co2p, 2017-08-15:
@guysoft ok, I can do a test run on both pi 1 and 2 tomorrow and see what happens, for future reference (and to see if I just didn't read the instructions last time I tried).

guysoft, 2017-09-26:
Hey, did you do this in the end? Closing if no reply

bmsimmons01, 2018-04-03:
I'm running 2018-03-15_2018-03-13-fullpageos-stretch-lite-0.9.0 from the Nightlies, and I'm experiencing the same black screen with white square in the bottom right corner as jandeman023 reported last year. Was there any resolution or suggestions on this?

guysoft, 2018-04-03:
Can you vnc and see if you see the same thing? I want to isolate if its a display issue or the browser is crashing,  or something else.

bmsimmons01, 2018-04-04:
When I VNC'd in I saw only a black screen (I think the white square in the bottom right-hand corner may have been part of a cursor. When I ssh'd into the pi I saw the same. I ran ps aux and found that Chromium is still running (attached output).
[pi_ps_aux_output.txt](https://github.com/guysoft/FullPageOS/files/1877855/pi_ps_aux_output.txt)

guysoft, 2018-04-04:
Ok, can you try disabling the GPU acceleration?

bmsimmons01, 2018-04-05:
I tried adding the --disable-gpu flag in the start_chromium_browser script, same issue (I also turned off in Chromium's settings page). I tried raspi-config > Advanced Options > GL Driver > Legacy. I'm loading a CSS / Javascript slider, very similar to the original poster (jandeman023). The slide begins but goes black after the second image. This may be a Chromium issue.

guysoft, 2018-04-05:
It may be. Or at least they might have the knowledge of what is going on. I am not really sure how to debug it. You may want to try asking on the chromium issue tracker:
https://bugs.chromium.org/p/chromium/issues/list

guysoft, 2018-07-22:
Closing due to no reply

closed 2018-07-22 by guysoft