guysoft/FullPageOS #124: Cannot Resolve local DNS address

conrad9900, 2017-04-19:
#### What were you doing?

Trying to get a local address to open on startup

#### What did you expect to happen?

For my page oms.profile.local to open

#### What happened instead?

Google homepage opens

#### Was there an error message displayed? What did it say?
No Error

#### Version of FullPageOS?

Downloaded 2 days ago, no octopi_version file in my /etc

[Can be found in /etc/octopi_version ALWAYS INCLUDE.]

#### Screenshot(s) showing the problem:

fullpageos-network settings:

auto eth0:1
iface eth0:1 inet dhcp
domain profile.local
dns-nameservers 192.168.0.10
#address 192.168.0.102
#netmask 255.255.255.0
#broadcast 192.168.0.210

guysoft, 2017-04-19:
It should be ``Fix /etc/fullpageos_version``, fixed now.

This is a raspbian issue, not a fullpageos. one.
Does this work on raspbian without FullPageOS installed?

guysoft, 2017-08-11:
Closing if no answer