guysoft/FullPageOS #127: How can I automatically refresh a page every X minutes?

cesarzapatawfm, 2017-04-30:
I looked around the distro and I could not find a good way to make Chromium  refresh the page every x minutes. 

Any scripts or will the new version have this capability? 

Not using FullPageDashboard. 
Using version 6

guysoft, 2017-05-10:
You didn't find one because there is none. 
Also FullPageDashboard, the app that handles the refresh on FullPageOS is not maintained anymore. You could modify its source directly or use another dashboard app.

danielhampton, 2017-07-30:
Two ways, if you're just showing a html page then either

<script type="text/javascript">
  setTimeout(function () { location.reload(true); }, 30000); // 30 seconds
</script>

or via a meta tag

<meta http-equiv="Refresh" content="30">

guysoft, 2017-08-11:
Closing if no answer

tdusmoha, 2017-10-27:
the simplest way I found is to schedule a cronjob to execute the refresh script in /home/pi/scripts. works fine for me..

benyanke, 2018-01-18:
The best way to do this would be:

1) run `crontab -e`
2) Insert the following lines into the file:

```sh
# Refresh the page every 5 minutes, to keep running well
*/5 * * * * /home/pi/scripts/refresh &> /dev/null
```

cesarzapatawfm, 2018-01-18:
Thanks guys.

closed 2018-01-18 by cesarzapatawfm

rettroo, 2019-03-11:
Need some help here... My Problem is that i have always the same homepage. I change them a few times but always the page from day1 is showing. even if i restart the Raspberry (why??). Now i tried to save the Line from above in cronjob. Tried so much but will not working....
now i have this in crone:
5 * * * * /home/pi/scripts/refresh 

i am not sure if i need the "*/" and the "&> /dev/null" but i try also with them!

any ideas? i use the latest nightly version

scfigg, 2021-02-23:
Ive installed the crontab per documentation here:
# Refresh the page every 5 minutes, to keep running well
*/5 * * * * /home/pi/scripts/safe_refresh &> /dev/null

However, it's not refeshing from what I can tell. Am I doing something else wrong? Do I need to enable anything else to get it to work? I can see the scripts, so I know I'm using a build with them in it.

Thanks

guysoft, 2021-03-01:
try sending the output from ``/dev/null`` to ``/tmp.refresh.log`` and see if that is outputting anything. 
Also - by documentation you mean this issue right?

WDazed, 2022-05-10:
Hi,

I've run `crontab -e`

I've then added in 
```
# Refresh the page every 5 minutes, to keep running well
*/5 * * * * /home/pi/scripts/refresh &> /dev/null
```

Unfortunately it's not refreshing the browser, is there something I've missed? 

I originally came here: https://github.com/guysoft/FullPageOS

Selected the link to the mirror (http://unofficialpi.org/Distros/FullPageOS/) and then downloaded the zip...
[image]

Any help would be greatly appreciated, thank you!

WDazed, 2022-05-23:
Ok, so this was doing my absolute head in. Thanks to @George-Britton I finally have it working!!

We used:

> cd /home/pi/scripts

> sudo chmod -R 777 /home

> cp safe_refresh /home/safe_refresh

> cd /home

> crontab -e

then added in:

> */5 * * * * /home/safe_refresh &> /dev/null

blastbeng, 2022-11-09:
Sorry to comment an old thread, but i still have an issue

Okay for refreshing, the script under /home/pi/script/refresh is working good

But i need to go back to the main page defined in /boot/fullpageos.txt

I have some user interaction on my dashboard, i want to go to the main page every hour because the user can change the tabs.
How to don that? Refresh is not enough in my case

actually i am doing 

`0 */1 * * * /usr/bin/killall /usr/lib/chromium-browser/chromium-browser`

but it is a dirty solution.

ethaniel, 2023-12-13:
> but it is a dirty solution.

Actually not, it cleans out any possible memory leak issues.

guysoft, 2024-01-15:
It could be added to the FAQ wiki page in this repo if you want to add it :)

pengytek, 2024-11-18:
> Ok, so this was doing my absolute head in. Thanks to @George-Britton I finally have it working!!
> 
> We used:
> 
> `cd /home/pi/scripts`
> 
> `sudo chmod -R 777 /home`
> 
> `cp safe_refresh /home/safe_refresh`
> 
> `cd /home`
> 
> `crontab -e`
> 
> then added in:
> 
> ` */5 * * * * /home/safe_refresh &> /dev/null`

I used these commands but the it never refreshes. The log file says "/bin/sh: 1: /home/safe_refresh: Permission denied"

tapio80, 2025-03-02:
I made simple Chrome extension which refreshes web page every 5 minutes.
https://github.com/tapio80/Refresh-Page

JAUBERT1977, 2025-05-10:
### Hi,

### I had the same problem on my kiok using FullPageOS, I solved the problem using  @pengytek  and @ethaniel  solutions. 

### cd /home/pi/scripts
### sudo chmod -R 777 /home
### cp safe_refresh /home/safe_refresh
### cd /home
### crontab -e
------------------------
### then added in:
------------------------
### */5 * * * * /home/safe_refresh &> /dev/null
-----------------------
### Save file
-----------------------

### chmod 755 /home/safe_refresh

### Thanks a lot.