guysoft/FullPageOS #128: New TV, black screen.

mgjuhler, 2017-05-27:
Today i have bought a new TV for use with my pi and FullPageOS.
I put my pi into my new tv and power it up, there comes this "colored" boot thing and after that nothing happens, the screen is black. I do not get any mouse or similar response when using keyboard and mouse. However, I can ssh to it and use it as i must without picture on the tv.
If I put my pi in my old tv then there is no problem.

I have checked the settings on the new TV and have been through it all without luck. The new TV does not write "no signal" unless i remove the hdmi cable from it.

I really hope that someone can help me :(

Build: fullpageos-jessie-lite-0.7.0_RC1 (Nightly)
Pi version: 3 B

New TV: philips 49PFS5301
Old TV: samsung UE55H8005SQ

mgjuhler, 2017-05-30:
Any help?

SManAT, 2017-05-31:
just a quick shot, i have the same problem on a normal LED Monitor
i tried the option hdmi_safe=1 in the config file (hope syntax is correct)
after that i got the OS on the screen here. But it hangs at logon screen ....
more in future ;)

SManAT, 2017-06-01:
Other approach that worked.
Lightdm seems to be the problem, so i used another manager
```
sudo apt-get install slim
sudo dpkg-reconfigure slim
```
now i can login to the system, but right now, nothing more ....

mgjuhler, 2017-06-10:
Well.. now i have tryed with "hdmi_safe=1"  and nothing changed.. so i tryed "2017-03-30_2017-03-02-fullpageos-jessie-lite-0.7.0_RC2" and still no changes.. when i run "2017-04-10-raspbian-jessie" on it works fine. So its a FullPageOS problem. i really need som kind of help to fix this problem :(

hlandao, 2017-06-13:
I have the same issue

tailorvj, 2017-06-29:
I've seen this happen with low voltage power input. Try to feed the Pi with at least 2V or even 2.5V power input. A lot of USB power outputs and phone chargers only provide 1A

guysoft, 2017-06-30:
please use RC3. Which is the latest

muoi, 2017-07-13:
Just had the same issue, it works on one monitor but not the other (both Dell). Although first boots works on both.

rbened, 2017-10-10:
Having the same issues. Ruled out the possibility being low voltage, tried with different versions of the OS, including the lastest.

Works fine on the old TV (LG43UH661V), the new TV (LG43UJ670V) only shows the 'color logo' and with a fresh install briefly shows the initial boot screen.

guysoft, 2017-10-10:
If you see the color screen it means the pi is not booting and it should not be affected by what you plug into the HDMI

rbened, 2017-10-11:
The color screen shows only briefly, not permanent. I'll try to make a video later to show what happens exactly.

guysoft, 2017-10-11:
Does the pi boot? Do you see the green LED flashing after that? If so then its clearly an HDMI issue.

muoi, 2017-10-11:
I ended up using another distro and have been since my last comment and it works fine on all monitors. There's a problem with this distro for some monitors I believe.

guysoft, 2017-10-11:
Can you link to that distro? The way to solve this is finding out what is difference between the two distros.

Gun-neR, 2017-11-07:
I just downloaded the latest version and installed after setting my WiFi.  on the TV/monitor I watched it boot, resize the SD, run through the colour splash and boot process... then blank.

Now whenever i power the RPi the monitor indicates a signal, but screen stays black ... then after about 20 seconds, the monitor switches to sleep as it looses the signal.

I have confirmed this on two different monitors, the 1st a 1280x1024 Monitor and the 2nd one a full 1920x1080.

I can SSH in and even see what I probably should see on the TV/monitor when I set my browser to http://<MyPiIP>/welcome/

Running on an RPi Zero W... yes I know it is slow, but it is all I have right now... and should still display something.

guysoft, 2017-11-07:
I tried a rpi 1 for the sake it, it didn't work.

Gun-neR, 2017-11-07:
That reminded me, I did have a RPi Model B v2.0. So I jammed the micro SD into an adapter and powered it up... and, same reaction in the monitor (very brief flash of the colour swatch, then no signal sleep mode), but I can SSH and also see it on the browser.  Well actually, it seems  I can see the welcome page, but when I go to just the IP, I get a placeholder page, despite the fact that I put `http://goggle.ca` in the `fullpageos.txt` file, so I might be doing something wrong there?... but still having nothing on the monitor is another issue.

regand02, 2017-11-07:
I have a similar problem with a Pi 2. With both 0.8.1 and 0.9.0, there is a black screen on every boot. Network is wired DHCP that admittedly requires a proxy, but I've tried it with network in and out and proxies configured and not. I can SSH and VNC in, in addition to being connected to the monitor, but nothing.

`ps aux` shows no chromium instances running at all, and run_onepageos seems to be running in a loop - the if statement seems to return a HTTP 400...

Gun-neR, 2017-11-08:
OK, I finally got a visible screen... looks a little strange though.  I think It started working after re- downloading and flashing 2017-10-22_2017-09-07-fullpageos-stretch-lite-0.8.1

But I did also "activate" force HDMI-1 or Mode-1 or something??  I don't want to pull the SD card, again,  just to see what I did, and I would have to since WiFi is no longer working (Yes, I also posted in proper topic).

Anyhow... is this what it is supposed to look like?

[image]

Gun-neR, 2017-11-08:
OK, got the WiFi working (one to many # removed)... and somehow that also changed the screen layout??

I guess now this is too zoomed in?

[image]

I guess I will keep fiddling with it :)

guysoft, 2018-09-16:
Another option to help this could be ``hdmi_boost`` config parameter:
https://www.raspberrypi.org/documentation/configuration/config-txt/video.md

Also, should I close this? it seems there is no specific issue here anymore

guysoft, 2018-09-16:
Ok, one more thing that might fix it is using the no-acceleration version.
nightly currently at: 
http://unofficialpi.org/Distros/FullPageOS/no-acceleration/nightly/

KiwiKilian, 2018-11-22:
We also had a black screen with only the cursor in the lower right corner visible as a white point. The "No Accelleration" Version fixed this problem. Thank you!

closed 2019-05-02 by mgjuhler

guysoft, 2019-05-03:
@mgjuhler Did the new non-acceleration RC fix your issue?