guysoft/FullPageOS #129: Keys are incorrect. @ and "

FatManLovesPi, 2017-06-01:
Does anyone know how to fix the @ and "? They are showing up reversed, so when I try to type @, I get the " symbol and vice versa.

Thanks

rawlk, 2017-06-30:
use "sudo raspi-config" to set the Localisation Options:
(assuming you're in the us) 
Change Locale:
  -> en_us.utf-8 utf-8
  -> Default locale for system: en_US.UTF-8
Change Keyboard Layout:
 -> Generic 104-key PC
 -> English (US)
 -> No AltGr key
 -> No compose key
 -> (your choice)

After a reboot, this fixed it for me.

While you're in there, might as well update your password and resize the root partition if you haven't already.

guysoft, 2017-08-11:
Closing if no answer (also thanks @rawlk for detailed instructions)

FatManLovesPi, 2017-08-17:
@rawlk Thanks for that. I ended up changing the default keyboard file to "us"

sudo nano /etc/default/keyboard
Change gb to us

closed 2017-08-17 by FatManLovesPi