guysoft/FullPageOS #134: FullPageOS 0.7.0 RC3

guysoft, 2017-06-25:
FullPageOS 0.7 second release candidate.

Image:
http://docstech.net/FullPageOS/2017-06-24_2017-06-21-fullpageos-jessie-lite-0.7.0RC3.zip

md5:
http://docstech.net/FullPageOS/2017-06-24_2017-06-21-fullpageos-jessie-lite-0.7.0RC3.zip.md5

The main change from RC2 is fixing of the GPU settings (that caused RaspberryPi to freeze on start of the GUI). RPi foundation changed the way the driver is loaded and that caused it.

If no one says this image is a problematic, this time I'll just release it, this time I mean it.

yannh, 2017-06-28:
Hi @guysoft , would you know if https://github.com/guysoft/FullPageOS/issues/66  is fixed? We tried 0.7 (maybe an older build) but had to revert to 0.6 because of that... It' d be fantastic if it made it to 0.7, I' m eyeing the hardware acceleration ;)

Thanks so much for the work!

guysoft, 2017-06-28:
Try 0.7.0RC3 released last week. It went a long way since. Would be really surprised if its still there

yannh, 2017-06-28:
Works like a charm, and hardware acceleration makes things much smoother! :+1:

guysoft, 2017-06-30:
Released, finally
https://github.com/guysoft/FullPageOS/releases/tag/0.7.0

closed 2017-06-30 by guysoft