guysoft/FullPageOS #135: YouTube Embed = Aw, Snap!

felunka, 2017-06-25:
I was embedding a YouTube Video and every time the page loads I get the YouTube loading animation and seconds after that the Chromium Error Page Aw, Snap! Something went wrong while displaying this Page. I also tried using the old XML embed to enforce Flash.

I am using the newest Version (2017-06-24_2017-06-21-fullpageos-jessie-lite-0.7.0RC3).

guysoft, 2017-06-25:
Does that happen in RC1? because it sounds like a duplicate of #29

felunka, 2017-06-25:
I have seen the issue #29 . I have tried installing a different Version but had the same Problem. I tried to embed Ustream now but got the same problem.

guysoft, 2017-06-25:
Link for RC1 download is at #92

felunka, 2017-06-25:
Thanks. I will try that now

felunka, 2017-06-25:
It works now, but it needs to rebuffer all the time. Is there a way to improve that?

guysoft, 2017-06-25:
Ok, seems like a regression in RC3. Do you have a link to the video?
Buffering means your internet connection is too slow

felunka, 2017-06-25:
I started fast.com on the Pi. It said 89mbps. Link to the Video: https://www.youtube.com/watch?v=RtU_mdL2vBM
To be honest: it crashed again and is not working anymore. I have no idear why

yannh, 2017-06-30:
Hi there, I can confirm I am having a similar issue. I have some other Pis setup with 0.6.0, it works, but it would probably be nicer with hardware acceleratoin (maybe that' s what' s causing it to crash?)

guysoft, 2017-06-30:
@yannh Yes its the GPU, that is konwn

stappersg, 2020-10-20:
I think this duplicate of #86 should be closed to prevent drain energy upon wading the issues.

guysoft, 2020-10-26:
Merging as duplicate of #86

closed 2020-10-26 by guysoft