guysoft/FullPageOS #137: Not booting

MattiasTR, 2017-07-04:
#### What were you doing?
I wanted to change the rotation of the screen by 90 degrees. I entered "display_rotate=1" in /boot/config.txt.

#### What did you expect to happen?
The Raspberry Pi 2 Model B should boot with the screen rotated 90 degrees

#### What happened instead?
The activity LED blinked a few times at power on, then nothing happens. The monitor leaves sleep mode, but the screen is black. No debug output is coming on the serial interface.

#### Was there an error message displayed? What did it say?
No error messages, just a blank black screen.

#### Version of FullPageOS?

[Can be found in /etc/fullpageos_version ALWAYS INCLUDE.]
2017-06-24_2017-06-21-fullpageos-jessie-lite-0.7.0.zip

I have changed the memory split to give the GPU both 128 and 256mb, but there was no difference. Commenting the "display_rotate=1" in the config.txt makes the RPi boot normally. Tested on both RPi 2 and RPi 3 with the same result.

Rotating the screen works on a NOOBS install of Raspbian.

guysoft, 2017-07-04:
Try booting normally and removing the GPU acceleration first as described
in the release notes of 0.7.0

MattiasTR, 2017-07-05:
Where can I find the release notes for 0.7.0?

guysoft, 2017-07-05:
In the release tab:
https://github.com/guysoft/FullPageOS/releases/tag/0.7.0

MattiasTR, 2017-07-05:
Thank you.

I tested turning off the GPU acceleratin and verified that it was off in Chromium, then rotated the display and the RPi booted as intended with the screen rotated.

Do you want me to close this issue with the resolution to turn off GPU acceleration or leave it open until a future release fixes it?

guysoft, 2017-07-07:
Its a Raspbian issue. Their GPU driver apparently doesn't support rotation.
This bug should be reported to the raspberrypi linux kernel at:
https://github.com/raspberrypi/linux/issues

And the driver you need to report about is called vc4-kms-v3d

guysoft, 2017-07-23:
Huh, looks like they are pushing fixes 🥇

guysoft, 2017-09-26:
Fixes should be in 0.8.0 nightly, you are welcome to test and report back:
http://unofficialpi.org/Distros/FullPageOS/2017-06-24_2017-06-21-fullpageos-jessie-lite-0.7.0.zip
http://unofficialpi.org/Distros/FullPageOS/2017-06-24_2017-06-21-fullpageos-jessie-lite-0.7.0.zip.md5

MattiasTR, 2017-10-02:
I tested the nightly build found hereon a RPi3:
[http://unofficialpi.org/Distros/FullPageOS/nightly/2017-09-18_2017-09-07-fullpageos-stretch-lite-0.8.zip]
[http://unofficialpi.org/Distros/FullPageOS/nightly/2017-09-18_2017-09-07-fullpageos-stretch-lite-0.8.zip.md5]

I added the line "display_rotate=1" in /boot/config.txt

The first cold boot (tested twice, reimaged the SD-card in between) the display is rotated until it automatically reboots, then it ignores the setting in the config.txt on all following cold and warm reboots.

The changing of slides in the default presentation seems to be very slow as well, it looks like it is not using the HW acceleration.

guysoft, 2017-10-02:
You can test if GPU acceleration is used by:
1. Press ctrl+tab to open a new window
2. Type ``chrome://gpu``

You could post a reply to the kernel team I guess.

conrad9900, 2017-11-01:
I had a look through the release notes but i am finding it difficult to find a simple command to diable GPU acc. Does someone have a command?

guysoft, 2017-11-01:
Only via ``raspi-config``. There is no command.  I copied the source code from raspi-config in order to get it to work in the first place.

rlucian, 2017-11-22:
I've read on this thoroughly but it's not clear whether HW acc is going to be available on rotated desktops.
Right now display-rotate=1 works but only after disabling hardware acceleration, in 0.8.1
Is there any work in progress on having acceleration functional on rotated screens?

guysoft, 2017-11-22:
There was a commit to the raspberrypi kernel regarding this issue two month ago. You can try updating the kernel and see if it fixes it.

Jaxel, 2017-11-22:
Where do I disable GPU acceleration in raspi-config?

guysoft, 2017-11-23:
Second result on google for "raspi-cofig gpi accelerarion" dude: https://docs.eltechs.com/install-and-configure-exagear-desktop/hardware-graphics-acceleration-on-rpi

Jaxel, 2017-11-23:
Okay, I got the rotation working using `display_rotate=1`... and its much better.

However, the touch screen itself is off. How do i fix this?

rlucian, 2017-11-23:
try this: https://www.raspberrypi.org/forums/viewtopic.php?t=120793

more precisely, in /boot/config.txt see below the comment on rotating both image and touch interface, didn't test to see if only a value of 2 works, or 1 (90 degrees) as well.
```
display_rotate=0 Normal
display_rotate=1 90 degrees
display_rotate=2 180 degrees
# NOTE: You can rotate both the image and touch interface 180º by entering lcd_rotate=2 instead
display_rotate=3 270 degrees
display_rotate=0x10000 horizontal flip
display_rotate=0x20000 vertical flip
```

rlucian, 2017-11-23:
After updating kernel with raspi-update, webgl acceleration works even with rotation, but only on FAKE KMS setting.

```
Graphics Feature Status
Canvas: Hardware accelerated
CheckerImaging: Disabled
Flash: Hardware accelerated
Flash Stage3D: Hardware accelerated
Flash Stage3D Baseline profile: Hardware accelerated
Compositing: Software only. Hardware acceleration disabled
Multiple Raster Threads: Enabled
Native GpuMemoryBuffers: Software only. Hardware acceleration disabled
Rasterization: Software only. Hardware acceleration disabled
Video Decode: Software only, hardware acceleration unavailable
Video Encode: Software only, hardware acceleration unavailable
WebGL: Hardware accelerated
WebGL2: Hardware accelerated
```

benyanke, 2018-01-18:
I was able to get system-wide screen rotation working. Since it was mentioned but not explicitly  documented above, I will document the exact steps here:

1) sudo raspi-config
2) 7 - Advanced Options
3) A7 - GL Driver
4) G2 - GL (Fake KMS)
5) Reboot

This should allow you to rotate the screen.

evandrobts, 2018-02-04:
Thanks benyanke! It worked perfectly. Im using fullpageos-stretch-lite-0.8.1 with RPi 3 Model B.

guysoft, 2018-02-05:
Seems like issue is solved closing.

closed 2018-02-05 by guysoft