guysoft/FullPageOS #139: Virtual Keyboard?

Morphy99, 2017-07-21:
How do I enable a virtual keyboard for my touchscreen display?

I've installed matchbox-keyboard and ran it as a daemon however it's not appearing?

guysoft, 2017-07-23:
Well, for a start running ``DISPLAY=:0 matchbox-keyboard`` from ssh works.
Running it as demon does not.
You could add it to the ``scripts/run_onepageos`` script. Or ``scripts/start_gui`` in the nightly builds

guysoft, 2017-08-11:
Closing if no answer

Morphy99, 2017-08-11:
Sorry haven't had time to test as yet as my touchscreen is with a friend.

remi-grumeau, 2017-08-11:
What about adding an extension for that?

https://chrome.google.com/webstore/detail/virtual-keyboard/pflmllfnnabikmfkkaddkoolinlfninn

Remi

davidus-sk, 2017-12-05:
Did anyone find a solution? The chrome extension quoted above does not work with dropdown/search/filter input fields; when you type, nothing shows up and no filtering happens. The matchbox keyboard inserted into any of the files mentioned by @guysoft creates a tiny, almost unusable keyboard at the bottom of the screen. Funnily enough, when I run the matchbox command by hand after the web app loads then the keyboard loads up fine with the correct size.

Ideal solution would be an onscreen keyboard which appears only when an input field is activated by touch (like the mentioned extension, but working on all types of input fields).

nextlevelkid, 2018-04-22:
Is there anyway i can add the virtual-keyboard extension from chromium and have it start on boot. ? 
It seems theres ways to call download extension with --apps-gallery-url or with somehow loading the ID of the extension and or the dir of the extension.. Can this be done somehow in <start_chromium_browser>
i got matchbox running inserting it into <start_gui> but it makes the browser page small because it takes up part of the layout on the screen. Matchbox doesnt seem to have a auto-hide feature and florence doesnt seem to want to work well. its buggy and doesnt pop up over the top of the fullscreen chromium. any solution for that would at least be a start. It seems either --fullscreen or --kiosk is not allowing anything to pop up over the top of browser window?
(* i got florence virtual keyboard working (behind chromium) by making a keyboard.sh script and adding it to the run_fullpageos file. ) but again its really clunky and glitchy
So overall it would seem like best idea to get the onboard chromium --virtual-keyboard and --float-virtual-keyboard working somehow and or be able to have the virtual keyboard from webstore preloaded and activated in chromium as an extension. Then it seems there wouldnt need to be any hacks to make keyboard work. just inside already running chromium. 
Any feedback or thoughts would be greatly appreciated. i really want to use this setup for our kiosk. 
Great works on this btw. it seems to be a perfect solution for us. 
Hope to get this solved.

Linkek, 2023-10-07:
I had the same issues. So I created this project to be able to trigger matchbox-keyboard from a Chrome extension. 

https://github.com/Linkek/keyboardTriggerServer

Thought some people might think it's somewhat useful.