guysoft/FullPageOS #140: Trying to Calibrate a touchscreen..Requires XServer

conrad9900, 2017-07-24:
#### What were you doing?
Trying to calibrate a touchscreen as it is not correctly set.
https://www.raspberrypi.org/forums/viewtopic.php?f=91&t=78805

#### What did you expect to happen?
xserver launches and shows a calibration screen

#### What happened instead?
XServer Did not start

#### Was there an error message displayed? What did it say?
sudo startx
xauth:  file /root/.Xauthority does not exist
X.Org X Server 1.17.2
Release Date: 2015-06-16
X Protocol Version 11, Revision 0
Build Operating System: Linux 3.8.11-2-ARCH armv7l Raspbian
Current Operating System: Linux fullpageos 4.4.11-v7+ #888 SMP Mon May 23 20:10:                                                                                                                                                             33 BST 2016 armv7l
Kernel command line: dma.dmachans=0x7f35 bcm2708_fb.fbwidth=800 bcm2708_fb.fbhei                                                                                                                                                             ght=480 bcm2709.boardrev=0xa01041 bcm2709.serial=0xb6d794a9 smsc95xx.macaddr=B8:                                                                                                                                                             27:EB:D7:94:A9 bcm2708_fb.fbswap=1 bcm2709.uart_clock=3000000 bcm2709.disk_led_g                                                                                                                                                             pio=47 bcm2709.disk_led_active_low=0 vc_mem.mem_base=0x3dc00000 vc_mem.mem_size=                                                                                                                                                             0x3f000000  dwc_otg.lpm_enable=0 console=ttyAMA0,115200 console=tty1 root=/dev/m                                                                                                                                                             mcblk0p2 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait
Build Date: 11 January 2016  01:32:44PM
xorg-server 2:1.17.2-1+rpi1 (http://www.debian.org/support)
Current version of pixman: 0.33.3
        Before reporting problems, check http://wiki.x.org
        to make sure that you have the latest version.
Markers: (--) probed, (**) from config file, (==) default setting,
        (++) from command line, (!!) notice, (II) informational,
        (WW) warning, (EE) error, (NI) not implemented, (??) unknown.
(==) Log file: "/var/log/Xorg.1.log", Time: Thu Jul 20 14:28:45 2017
(==) Using system config directory "/usr/share/X11/xorg.conf.d"
^[[A^[[B^Cxinit: connection to X server lost

waiting for X server to shut down (II) Server terminated successfully (0). Closi                                                                                                                                                             ng log file.

xinit: unexpected signal 2

#### Version of FullPageOS?
Newest

[Can be found in /etc/fullpageos_version ALWAYS INCLUDE.]

#### Screenshot(s) showing the problem:

[If applicable. Always include if unsure or reporting UI issues.]

guysoft, 2017-07-24:
X is already starting, you are getting an error ``xauth: file /root/.Xauthority does not exist`` Because root has no permission to access the existing X.
Please run with ``DISPLAY=:0 xinput_calibrator`` and it should run

conrad9900, 2017-07-24:
will test it now

conrad9900, 2017-07-24:
That worked perfect, what is the quickest method for resolution change?

guysoft, 2017-08-11:
Xrandr probably.

closed 2017-08-11 by guysoft