guysoft/FullPageOS #141: Editing the fullpageos.txt from Pi

sng1, 2017-07-26:
Hi,  Is there a way to edit direct the fullpageos.txt directly from the Pi while it's running?

ctrl + alt + F1

Where do I go from there to get to that file and edit it to change the URL?

Thanks

guysoft, 2017-07-26:
You can if you open a terminal and edit it.
You can't via the browser, that's why we ship with FullPageDashboard which
lets you add pages.

sng1, 2017-07-26:
FullPageDashboard is locally on the Pi?

What's the path to find  fullpageos.txt from the command line?

sng1, 2017-07-26:
I guess nano fullpageos.txt   will do the trick!!!    I'll have to play around with it!   Thanks

guysoft, 2017-07-26:
It says so in the readme under the feature list:
https://github.com/guysoft/FullPageOS/blob/devel/README.rst#features

closed 2017-07-26 by guysoft