guysoft/FullPageOS #142: Rotate Time Not Working

sng1, 2017-07-27:
I have 2 pages I'd like to flip through   I can set both up on the Dashboard   I reboot the Pi but nothing is happening.  I see it switching on the Dashboard but not on the Pi

guysoft, 2017-07-27:
How do you see it switch not on the raspberrypi?

sng1, 2017-07-27:
Trough the HDMI out to a TV monitor.   On the Dashboard http://fullpageos.local/FullPageDashboard/  from Chrome running MacOS I have 2 tabs with 2 different URLs.   On the Pi's HDMI signal I only see one URL getting displayed.  

Just to be clear the Dashboard is only for remote management? 
Basically all the settings get done from the Dashboard then get transfered remotely to the Pi?

Is there a way to put the 2 URLs straight into the fullpageos.txt with the rotation time right on the SD card?

I want to use FullPageOS 2 display 2 websites in rotation and the websites will never change it's for a permanent installation basically set and forget, they get dynamically changed from a server.  I don't even need to SSH into the Pi.  If it crashes I would simply manually restart it even though I can easily SSH on my current setup.

guysoft, 2017-07-27:
1. Refresh the page
2. No its also the display
3. Only on first biit in the fullpagedashboard.txt file.
4. Thats the default behavior

sng1, 2017-07-27:
1) So you're saying that I can have the page with the switching tabs right on the Pi's display, right from Chromium?

2) How do I shrink the the page so I can access the URL tabs?

3) Does FullPageOS works even when the Pi is not connected to a remote computer?

Not sure what you meant by the following
 3.**Only on first biit in the fullpagedashboard.txt file.**

guysoft, 2017-07-27:
1. Ah, you are actually editing the chromium tabs. 
I was talking about FullPageDashboard's in-webpage tabs. I don't know how that works, so I am not supporting it. You are welcome to document your own work.

2. Press control-T to get a new webpage with tabs
3. yes, but you will have no internet

sng1, 2017-07-27:
OK I think I got it   I built everything from the Pi with the + tab  I have 2 sites rotating every 120 seconds

1)  If I reboot the Pi will I lose those settings?   The idea is to setup a bunch of Pis the way it is now and deploy them to feed different monitors in a building.  All the Pis will have as far as connection is    Power, internet and HDMI to feed the TV(s)

2)  I use ApplePI-Baker to create my modified .iso so I can burn the SD cards with the backup iso I created.  Right now my prototype is working fine.  If I create a backup iso of the current SD card will all the data be copied as well? 

3)It doesn't seem to be fill screen I can see the pause button and the tabs name at the top.

Thanks

guysoft, 2017-07-27:
To 1, 2 and 3 - I have no idea, you are not using the supplied dashboard, so I can't know what exactly you did. Ask in a chromium forum.
You can disable the kiosk mode at boot by removing ``--kiosk`` from [/home/pi/scripts/start_chromium_browser](https://github.com/guysoft/FullPageOS/blob/devel/src/modules/fullpageos/filesystem/home/pi/scripts/start_chromium_browser). That's all I can help from here.

sng1, 2017-07-27:
I want to keep the Kiosk mode!

By "supplied dashboard" are we talking http://fullpageos.local/FullPageDashboard/ from the Pi itself on Chromium?  If so that's what I am using  I see at the top the cycling time and the pause button

guysoft, 2017-07-27:
Ok, so I don't understand what you aare doing with it. What haven't you
solved?

sng1, 2017-07-27:
OK so I got it to work.  Created a backup of the SD card on a desktop and burned that backup to a fresh SD card, put it back in the PI and both pages loaded like previously.   Only thing is that iso image is 63GB!!!  Only difference between the iso install and my backup is I added 2 URL.  Took me over 1 hour to burn with ApplePi Baker   Any idea why the file is so huge ?

guysoft, 2017-07-28:
Because the image expends to fill your sd card.
https://www.raspberrypi.org/forums/viewtopic.php?t=19468
I'm closing this issue. You keep changing topics, this isn't a support forum.

closed 2017-07-28 by guysoft

guysoft, 2017-07-28:
Regarding the shrinking, just came across this:
https://github.com/aoakley/cotswoldjam#raspbian-shrink

sng1, 2017-07-29:
Perfect thanks