guysoft/FullPageOS #144: This currently fails to build on Ubuntu 16.04

bhartvigsen, 2017-07-31:
#### What were you doing?
Building FullPageOS

#### What did you expect to happen?
Successful build after following instructions

#### What happened instead?
Build failure

#### Was there an error message displayed? What did it say?
Error: could not find image:
++++ previous_command='echo "Error: could not find image: $BASE_ZIP_IMG"'
++++ this_command='exit 1'
+++ exit 1
+ exit 1
+ exit_code=1
+ '[' 1 -ne 0 ']'
+ exit 1
#### Version of FullPageOS?

[Can be found in /etc/fullpageos_version ALWAYS INCLUDE.]
Nope: cat: /etc/fullpageos_version: No such file or directory

This project fails to build on Ubuntu 16.04.2 LTS. I followed the instructions in the README verbatim and got the above error. I had no errors or problems when setting up the dependencies, only when trying to run build_Dist

guysoft, 2017-07-31:
Build says "Error: could not find image: $BASE_ZIP_IMG"'
Did you place the latest raspbian-lite image in ``src/image``?

bhartvigsen, 2017-07-31:
I think I may have figured out the problem. In your instructions, it says to run this command:

    curl -J -O -L  http://downloads.raspberrypi.org/raspbian_latest

This places a file called 'raspbian_latest' in the image directory, however your script expects to locate a filename containing '-raspbian.zip' so by manually renaming the file the build is able to run as expected. My advice would be to update the README to change the command to:

    curl -J -L  http://downloads.raspberrypi.org/raspbian_latest > raspbian_latest-raspbian.zip

guysoft, 2017-07-31:
Yes I was just typing that and you figured it out before I sent the comment :)
Ideally I want to add a command from CustomPiOS that I wrote that downloads the latest image with the right name.
At the moment [its a function here](https://github.com/guysoft/CustomPiOS/blob/CustomPiOS/src/make_custom_pi_os#L32)

commit 197fba18 (2017-08-06): Fixes #144
closed 2017-08-06 by guysoft