guysoft/FullPageOS #146: FullPageOS booting to Debian 8 login screen.

cesarzapatawfm, 2017-08-09:
#### What were you doing?

nothing. My page has been working for weeks.

#### What did you expect to happen?
work as the other older verions of fullpageos
#### What happened instead?
instead of displaying my page, fullpageos is booting to a Debian 8 login screen. 

#### Was there an error message displayed? What did it say?
no error. I tried loggin with the pi login but it just reverts back to the Debian 8 screen.

#### Version of FullPageOS?

Could not get it as I had to reformat. 

#### Screenshot(s) showing the problem:

login similar to this one 
http://www.cnx-software.com/wp-content/uploads/2015/09/Orange_PI_Debian_8_Login.jpg

guysoft, 2017-08-09:
Is there a way to reproduce this?

guysoft, 2017-09-26:
Closing, no reply

closed 2017-09-26 by guysoft

guthubuser, 2017-10-09:
i can confirm this. this also happens to my installation. 
The debian screen appears and no change to revert. Where are the credentials stored? how is the autologin done?
I can login with root account and then the scipt continues

guysoft, 2017-10-09:
@guthubuser please use the nightly with debian 9.

guthubuser, 2017-10-09:
after a few reboots the same thing happened with debian 9. Can you tell me how the autologin is done? it is NOT in the /etc/lightdm/lightdm.conf 
[SeatDefaults] 
 autologin-user=pi
 autologin-user-timeout=0

guysoft, 2017-10-09:
This is the code pn build that handles it, if you upgrade it might get overwritten:
https://github.com/guysoft/CustomPiOS/blob/CustomPiOS/src/modules/gui/end_chroot_script

ianpowell2017, 2023-12-31:
I had the same issue.  I had created a new user and password overwriting the default 'pi' user.
I re-flashed the image using the default 'pi' user and it worked again.
I hope this helps someone