guysoft/FullPageOS #147: How to hide tab bar in Chromium?

trueaim1, 2017-08-09:
Installed version .7 (after having everything working in .6) and expected to see Chromium run without a tab bar as in version .6.  However, no matter how I modify start_chromium_browser, run_onepageos, fullscreen, etc, I can't get rid of the tab bar and '+' sign at the top of the Chromium window.  Can someone please point me in the right direction?  Thank you!

remi-grumeau, 2017-08-09:
--kiosk ?

Remi

trueaim1, 2017-08-09:
Thank you for the reply but I am using the --kiosk option/switch.  I will post my boot files when I get back to my computer.

trueaim1, 2017-08-10:
The contents of my boot files are as follows:
start_chromium_browser:
================
chromium-browser --kiosk --touch-events=enabled --disable-pinch --noerrdialogs --app=$(head -n 1 /boot/fullpageos.txt)

run_onepageos:
==========
#!/bin/bash
xset s off         # don't activate screensaver
# xset -dpms         # disable DPMS (Energy Star) features.
xset s noblank     # don't blank the video device

compton --backend glx --unredir-if-possible --glx-swap-method buffer-age --glx-no-stencil --paint-on-overlay --vsync opengl matchbox-window-manager &
matchbox-window-manager &
x11vnc -many -rfbauth /home/pi/.vnc/passwd &
while true
do
    if [ $(curl -sL -w "%{http_code}\\n" "http://localhost/FullPageDashboard" -o /dev/null) == "200" ] || grep -q disabled "/boot/check_for_httpd" ; then
      (sleep 15 ; /home/pi/scripts/fullscreen) &
      xdotool mousemove 9000 9000
      /home/pi/scripts/start_chromium_browser
    fi
    sleep 1
done

fullscreen
======
#!/bin/bash
export DISPLAY=:0
WID=$(xdotool search --onlyvisible --class chromium|head -1)
xdotool windowactivate ${WID}
xdotool key F11
xdotool mousemove 9000 9000

guysoft, 2017-08-10:
If you didn't change anything since the install there should be no bar. Can you post a photo?  I am getting the feeling you might be talking about the fullpagedashboard tab row.

trueaim1, 2017-08-10:
Yes, I believe I am talking about the fullpagedashboard tab row.  That is what I would like to remove.

And thank you for the nice distribution.  This will work great for our purposes once we figure out how to remove this tab row!

[image]

guysoft, 2017-08-10:
Then change thr url you want to show in ``/boot/fullpageos.txt`` . This is written in the README and is standard behavior.

closed 2017-08-10 by guysoft

trueaim1, 2017-08-10:
Yes, I understand that and have done that.  Here are the contents of /boot/fullpageos.txt 

# http://localhost/FullPageDashboard
http://jobportal.ourdomain.local/apply-home.html

It opens that page just fine but still displays the tab bar.

Thank you

trueaim1, 2017-08-10:
Update:
  I just got rid of the first line which I assumed would be interpreted as a comment and rebooted and all is well.  The page display without the tab bar.

Thanks again.