guysoft/FullPageOS #148: Booting to Kiosk mode but cycling between three Chromium tabs?

esaruoho, 2017-08-09:
Hi, I'm trying to figure out if I could get FullpageOS to boot up and start three Chromium tabs in Kiosk mode, but then actually switch between displaying tab1, tab2, tab3 at a given time? Like every minute, go from tab1 to tab2.. or tab2 to tab3.. or tab3 to tab1. kinda like a page-carousel. Would FullpageOS allow for this kind of behaviour?

guysoft, 2017-08-11:
No apart from FullPageDashboard there is no such functionality, you are welcome to research and report here. I just never had use for this.

esaruoho, 2017-08-11:
Unfortunately FullPageDashboard is kind of an oxymoron since it is not an actual full page. I'm not PHP savvy so please don't repeat your _suggestion_ of "stop using it or code it better"

closed 2017-08-11 by esaruoho

guysoft, 2017-08-12:
I'll would also accept a pull request of a dashboard that is coded better in a different language. FullPageDashboard is not maintained anymore by its developer.