guysoft/FullPageOS #149: How to schedule daily boots (every 24 hours)

esaruoho, 2017-08-11:
I have a Geckoboard dashboard running on FullPageOS - thing is that every few days, the whole setup will crash and refuse to continue. How do I make FullPageOS reboot every 24 hours (at midnight, for instance)?

guysoft, 2017-08-11:
1. What version of FullPageOS are you using?
2. You could try and disable GPU acceleration
3. You could set a cronjob

guysoft, 2017-09-26:
Hey, I've been testing the latest nightly of 0.8.0, and its MUCH more stable than 0.7.0, I also had "aww snap" crashes that stopped, suggest you try that.

digitalknk, 2018-01-24:
`sudo su`
`crontab -e`

add this line
`0 5 * * * /sbin/shutdown -r now`

save and exit

Now it will reboot everyday at 5AM

guysoft, 2018-01-24:
Closing due to no reply since September of OP, also 0.8.1 is really much more stable.

closed 2018-01-24 by guysoft

esaruoho, 2018-01-24:
@digitalknk thanks for this, i've set it to work and it works. thanks a lot
i guess i need to install 0.8.0 or something..

commit 5b837c86 (2015-11-28): Get build to work on Debian Jessie fixes #149