guysoft/FullPageOS #150: help installing FullPageOS on qemu

kamomil, 2017-08-13:
Hi , 
I built FullPageOS from the master branch,
I tried to run it in qemu.
I have no past experience working with qemu nor raspberry pi.
I followed the instructions from here https://www.raspberrypi.org/forums/viewtopic.php?f=29&t=37386 with two changes:
1. used the 2017-07-05-raspbian-jessie-lite.img that I built instead of the official rasbian
2. the link to download kernel-qemu http://xecdesign.com/downloads/linux-qemu/kernel-qemu is dead.
Instead , google lead me to https://github.com/dhruvvyas90/qemu-rpi-kernel , so I cloned it and used kernel-qemu-4.4.34-jessie as the kernel-qemu

Then I ran:
qemu-system-arm -kernel kernel-qemu -cpu arm1176 -m 256 -M versatilepb -no-reboot -serial stdio -append "root=/dev/sda2 panic=1" -hda ./2017-07-05-raspbian-jessie-lite.img -redir tcp:5022::22

The qemu poped and immediately closed

guysoft, 2017-08-13:
Hey,
1. Those instructions are out of date and don't work with Raspbian Jessie.
2. If you are using the versailepb machine then you need to have a different kernel compiled to run on it. The raspberrypi is a different machine and has a different kernel. I saw that there is a decently new ``

kamomil, 2017-08-13:
Hi,
Aren't those kernels in the link https://github.com/dhruvvyas90/qemu-rpi-kernel  specifically built for Rpi ? I don't have raspi2 on my installed qemu-system-arm

guysoft, 2017-08-13:
Ok so 
1. Yes, those kernels are out of date. Really out of date.
2. You might need to compile qemu from their master (or devel).

kamomil, 2017-08-13:
Hi,
I followed these instructions: http://blog.3mdeb.com/2015/12/30/emulate-rapberry-pi-2-in-qemu/ 
I compiled qemu locally and according to the instructions I took kernel7.img from the first partition of 2017-07-05-raspbian-jessie-lite.img , then when running the compiled qemu with "-M raspi2" and "-kernel kernel7.img" (the exact command was not saved in history)
a black window  popped and my computer got stuck. I had to power it off and on.

remi-grumeau, 2017-08-13:
> 30/emulate-rapberry-pi-2-in-qemu/

Hi list,

I installed qemu via brew (macOS). i have 2.9.0 installed.
i tried to boot raspbian jessie & latest fullpageOS using a kernel i got in
jessie img.
kernel file is called "kernel-qemu-4.4.34-jessie".
(as pointed out here : https://github.com/dhruvvyas90/qemu-rpi-kernel)

Qemu start, system starts but i end up in the terminal. I seems it fails
loading the kernel.
I face "a start job is running for dev-mmcdlk0p1.device" for 1m30.
Then i end in the terminal in Emergency mode.

I tried some stuffs using fsck but nothing works.
Neither jessie ou fullpageos loads.

Any assistance would be VERY appreciate to boot up a FullPageOS image!!

Remi

Remi

guysoft, 2017-08-20:
Hey,
So I tried checking out latest qemu master and boot. It goes further than the stable on Ubuntu. But it still gets stuck here. Screenshot attached.
[I followed this guide](https://raspberrypi.stackexchange.com/a/71172/7483), which uses the raspi machine. Here is my command to launch
```
~qemu/arm-softmmu/qemu-system-arm -M raspi2 -kernel ./kernel7.img -dtb ./bcm2709-rpi-2-b.dtb -sd /home/guy/workspace/custom_raspberrypi_os/AlarmPi/src/workspace/2017-08-16-alarmpi-stretch-lite-0.1.img -append "$(cat cmdline.txt | sed 's/quiet//g' | sed 's@init=/usr/lib/raspi-config
/init_resize.sh@@g' )"
```

It looks like chunks of the raspberrypi hardware are not emulated yet.
 
[image].

guysoft, 2017-09-27:
Hey, in the meantime, if you want to build a test environment, here is another method which lets you boot the pi over ethernet from a laptop:
https://github.com/anholt/linux/wiki/Raspberry-Pi-development-environment

closed 2017-12-31 by guysoft

guysoft, 2017-12-31:
Hey,
So I got this running the comment form [here](https://github.com/dhruvvyas90/qemu-rpi-kernel/issues/37#issuecomment-354533313), which direct to [this wiki entry](https://github.com/dhruvvyas90/qemu-rpi-kernel/wiki/Emulating-Jessie-image-with-4.x.xx-kernel).

I added to CustomPiOS [a script that lets you run any zip image in qemu](https://github.com/guysoft/CustomPiOS/blob/CustomPiOS/src/qemu_boot.sh). Note chromium is slow, really slow, because you are emulating arm instruction sets and there is no GPU acceleration. Another way would be to build FullPageOS on an intel system and run that with virtualization and not emulation.

Also, for FullPageOS you need to run the script twice because there is a reboot.

Example usage:
``./qemu_boot.sh /tmp/2017-10-22_2017-09-07-fullpageos-stretch-lite-0.8.1.zip``
(twice)
I also had to mount ``/dev/sdc1`` to ``/boot``after `ssh to localhost -p 5022`` 
[image]