guysoft/FullPageOS #152: Trying to display a Spiceworks dashboard

alanjm99, 2017-08-21:
[image]
[image]

Trying to display a webpage

[Please be as specific as possible here]
See image 2

#### What happened instead?
See image 1

#### Was there an error message displayed? What did it say?
None
#### Version of FullPageOS?
0.7.0

[Can be found in /etc/fullpageos_version ALWAYS INCLUDE.]

#### Screenshot(s) showing the problem:

[If applicable. Always include if unsure or reporting UI issues.]

guysoft, 2017-08-21:
Can you try editing the text file in ``/boot/fullpageos.txt``and not using the FullPageDashboard app?

alanjm99, 2017-08-21:
I am using the fullpageos file for the web address

pi@fullpageos:~ $ more /boot/fullpageos.txt
https://on.spiceworks.com/dashboard?timeframe=last_month

guysoft, 2017-08-21:
To possible solutions:
1. Looks like bad cache. cleaning it might solve it.
2. If the nightly works for you, I'd try 0.8.0 with stretch which has Chromium 60, it might solve that too (I still need to test it myself).

alanjm99, 2017-08-21:
Do i just download the nightly version and rebuild it...?

guysoft, 2018-02-16:
This issue should be solved in 0.8.1. Closing if no reply