guysoft/FullPageOS #153: Nightly 2017-08-16 black screen with cursor

ludagoo, 2017-08-23:
I have a stable version on on sd card and a nightly on a other one. the stable works great except it doesn't handle a web app with lots of animation well. I wanted to see if stretch would do any better. I can't get the nightly 2017-08-16-fullpageos-stretch-lite-0.8 to load the page. Any ideas?

guysoft, 2017-08-23:
I didn't get time to test it myself. And I wont for two weeks because I an flying tomorrow to Burning Man. 
Can you post the detailed explanation that is given in a new bug? its there for a reason.
Because I want to know:
1. what stable version are you using - because if its 0.6.0 its irrelevant. And 0.7.0 would rule out a GPU problem
2. A screenshot would help - is there a cursor on the top left of the screen?

guysoft, 2017-08-23:
Sounds like same behavior of https://github.com/guysoft/FullPageOS/issues/152#issuecomment-323745463

guysoft, 2017-08-24:
Ok, that nightly is messed up, 
It failed because of this bug:
https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=845720
rebuilding

ludagoo, 2017-08-24:
Where you able to kick off the build before leaving

guysoft, 2017-08-24:
Yes, should have just finished uploading. Check out the nightly folder

alanjm99, 2017-08-24:
It was uploading got to about 456MB now its disappered

ludagoo, 2017-08-24:
It's back, only 2/3 the size of the last build... downloading now... we'll see if it works...

alanjm99, 2017-08-24:
Its still uploading......

guysoft, 2017-08-24:
The suspense is terrible, I hope it will last.

ludagoo, 2017-08-24:
Indeed! 😂

ludagoo, 2017-08-24:
where'd it go... 😂

alanjm99, 2017-08-25:
Making some progress :) 

sudo apt-get update
sudo apt-get install dirmngr

sudo apt-get install chromium-browser

sudo apt-get purge lightdm
sudo apt-get update
sudo apt-get install lightdm
sudo dpkg-reconfigure lightdm

Now have a login screen

ludagoo, 2017-08-25:
Then what?

ludagoo, 2017-08-28:
I built the last commit  695dc37 and it all works great now.

closed 2017-08-28 by ludagoo

it-can, 2017-08-30:
@guysoft can you upload a new nightly? The current one is displaying a black screen with a cursor :-(

ludagoo, 2017-08-30:
@guysoft said:  "I wont for two weeks because I an flying tomorrow to Burning Man." six days ago

ludagoo, 2017-08-30:
@it-can here is my build for the latest commit... https://goo.gl/fbnhn2

it-can, 2017-08-31:
@ludagoo wifi doesn't seem to work :-(

alanjm99, 2017-08-31:
All I get is untrusted/certificate errors

ludagoo, 2017-08-31:
@it-can do you mean you downloaded and installed the image and can't get the pi to connect to the wifi? If so I haven't tried with wifi. I have it plugged in.

it-can, 2017-08-31:
@ludagoo yup