guysoft/FullPageOS #154: Fix clock?

Bortas, 2017-09-01:
This is going to sound excessively newbie like, I get that... I am. I've installed the most recent version today, I successfully edited the two documents to connect it to my wifi and launch the webpage of my choosing.

I can't figure out how to set system time. I can't figure out how to open system menu. I'm sorry I'm new and slow.

This project is excellent for those who don't understand diddly about linux or coding, outside of my HTML experience! Sorry for the dumb question.

guysoft, 2017-09-17:
You can use ``tzselect`` from commnadline to set the timezone.

guysoft, 2017-09-26:
Closing if no reply

Jaxel, 2017-11-22:
How do I access the command line?

guysoft, 2017-11-23:
@jaxel
1. You access it with ctrl+alt+F1 or via ssh
2. Please google before posting here.

closed 2017-11-23 by guysoft