guysoft/FullPageOS #157: Chromium doesn't use the full vertical space

Boscop, 2017-09-07:
Why does the default page look like this with FullPageOS 0.7.0, Rpi3?
[image]
Chromium doesn't use the full vertical space.

Staal-hb, 2017-11-03:
Same issue for me with 0.7.0, is there some way to upgrade to 0.9.0 from a already installed version?

If i use a other computer and connect to dashboard it looks correct.

i tried to upgrade chromium to v56 but there was no change.

guysoft, 2017-11-03:
You can try ``apt-get dist-upgrade``. But I am not supporting it and never
tried it. There is no uupgrade method,  suggest you flash 0.8.1 or 0.9.0

dalepmay, 2018-02-07:
Is this a normal issue? Is FullPageOS actually intended only for small screens like phones or tablets? I have installed versions 0.6 - 0.81 and every one shows the small unreadable view. If this is actually meant as a joke, well played sir.

guysoft, 2018-02-07:
@dalepmay Its a bug where the welcome page loads small in FullPageDashboard. Its fully functional apart from that. No one has managed to solve it so far.
If you don't like FullPageOS you dont need to use it. 
If you want to fix it the sources are available at https://github.com/amitdar/FullPageDashboard 
The welcome code is at https://github.com/tailorvj/FullPageOSWelcome

guysoft, 2019-06-18:
This has been resolved.

closed 2019-06-18 by guysoft