guysoft/FullPageOS #160: Resolution too low with browser-sync

Boscop, 2017-09-07:
My situation is, I want to auto reload a html file when it changes on disk (with another software writing the html file).
I used [this guide](http://thisdavej.com/beginners-guide-to-installing-node-js-on-a-raspberry-pi/#install-node) to install nodejs 8.4.0, npm 5.3.0. 
Then I installed browser-sync.
I modified `/boot/fullpageos.txt` to `http://localhost:3000` and then added this line to `scripts/run_onepageos` after x11vnc and before the while loop:
`(cd /home/pi/html; browser-sync start --no-notify --no-open -s -f './**/*.*') &`
That's my only modification.
It kinda works, whenever I modify the files it refreshes the page in the browser, but I get a very low resolution for the whole screen:
[image]
That's with this `index.html` in `/home/pi/html`:
```html
<!DOCTYPE html>
<html lang="en">
   <head>
      <title>foo</title>
      <meta charset="utf-8" />
   </head>
   <body>
<iframe src="http://www.youtube.com/embed/oHg5SJYRHA0?autoplay=1" width="640" height="400" frameborder="0" allowfullscreen></iframe>
   </body>
</html>
```

Why is the resolution so low?
How can I make use of the full resolution like before?
Where does it get the resolution setting from and why is it so low, just because I'm using browser-sync?

Boscop, 2017-09-07:
Ah, by repeating the process step by step I figured out that the monitor stays black already after I did `apt-get update && apt-get upgrade` and reboot after a fresh install of FullPageOS 0.7.0.
But why would the upgrade cause my monitor to stay black?

guysoft, 2017-09-26:
Similar report in #168 .
Suggest you try the 0.8.0 nightly

guysoft, 2018-02-16:
Closing if no reply

closed 2018-04-23 by guysoft