guysoft/FullPageOS #161: ftp

larrym, 2017-09-07:
#### What were you doing?

[Please be as specific as possible here]

#### What did you expect to happen?

#### What happened instead?

#### Was there an error message displayed? What did it say?

#### Version of FullPageOS?

[Can be found in /etc/fullpageos_version ALWAYS INCLUDE.]

#### Screenshot(s) showing the problem:

[If applicable. Always include if unsure or reporting UI issues.]

larrym, 2017-09-07:
Running version 2017-06-24_2017-06-21-fullpageos-jessie-lite-0.7.0, is it possible to transfer a file to the RPi via ftp or sftp?
If it is, what port and credentials?

guysoft, 2017-09-07:
Same as raspbian, sftp works. Ftp needs installation.

guysoft, 2017-09-26:
Closing, no description, no response.

closed 2017-09-26 by guysoft