guysoft/FullPageOS #164: Turn off monitor at night and turn it on again in the morning

cweagans, 2017-09-14:
I think it would be useful to have this option OOTB.

At some configurable time at night, run `tvservice -o` to turn off the display.

In the morning, `tvservice -p` will turn it back on again using preferred settings.

On my pi, `tvservice -p` isn't working for some reason, but I'm choosing to just reboot the pi first thing in the morning instead.

guysoft, 2017-09-14:
That can all be done in crontab. Just saying.

cweagans, 2017-09-14:
I know it _can_ be done there. I did it that way.

I'm suggesting that this would be a good thing to do OOTB.

guysoft, 2017-09-14:
No, because standard behavior is always on. Which is what most people want.
Adding time-based stuff requires you to also set the time zone.

cweagans, 2017-09-14:
Would you accept an opt-in way to do this through a config file or something? The energy savings is small, but with lots of displays, it will add up.

guysoft, 2017-09-15:
Yes, you are welcome to send a pull request

guysoft, 2017-09-17:
Closing if no reply

closed 2017-09-26 by guysoft