guysoft/FullPageOS #165: fullpageos does not respond

kamomil, 2017-09-18:
It tried installing Fullpageos on pi 2 model B that has no wireless. Since I'm too far from the router I configured it to use the wireless from my laptop, more or less as instructed [here](https://www.raspberrypi.org/forums/viewtopic.php?t=6997&p=87671) .

My issue is that the fullpageos doe not respond. It uploads and there is a tab with '+' sign that I guess should open a new tab when clicked, but it just does not respond.

guysoft, 2017-09-20:
Might be a cache issue, try to clear the cache.

kamomil, 2017-09-21:
you mean chromium's cache?

kamomil, 2017-09-23:
Hi, I'm not sure I understand how this os works.
Now, after booting it again , when I click the '+' I get a dialog for adding a new tab with url, then in the dialog I add 'huji' with url 'new.huji.ac.il' and then there is  a new 'huji' tab but the url is 'http://192.168.250.1/FullPageDashboard/#url' and clicking it does nothing.

I also removed all the files in:
**~/.cache/chromium/Default/Cache/ **
It seems to have no effect. 

another thing, not sure if it relates, but in the file **/boot/fullpageos-network.txt** I have:
                                                                                                                   
**
auto eth0:1                                                                                                        
iface eth0:1 inet static                                                                                           
  address 192.168.250.1                                                                                            
  netmask 255.255.255.0                                                                                            
  broadcast 192.168.250.255 
**

and in the file **/etc/network/interfaces** the last 3 lines are:
**
source /boot/fullpageos-network.txt                                                                                
                                                                                                                   
nameserver 208.67.222.222                                                                                          
nameserver 208.67.220.220    
**
and still the ip of eth0 on boot is not 192.168.250.1

thanks

guysoft, 2017-09-24:
If you are using FullPageOS nightly (you didn't provide a version, the fix in https://github.com/guysoft/CustomPiOS/commit/a3d4844f9b335fe0f60f0921d577d664a5f6c888 might fix this issue too, because ``clear_lighttpd_cache.service`` should have stopped working in the nightly.

guysoft, 2017-10-19:
Closing if no reply

closed 2018-02-16 by guysoft