guysoft/FullPageOS #166: Changed passwd and can't log in no

12nick12, 2017-09-20:
#### What were you doing?
1. Changed passwd on terminal
2. Changed passwd on ssh

#### What did you expect to happen?
1. I figured the password I changed it to would work over SSH
2. I figured the password I changed it to would work on the terminal

#### What happened instead?
1. Password doesn't work on ssh
2. Password doesn't work on terminal

#### Was there an error message displayed? What did it say?
No 

#### Version of FullPageOS?
0.7.0
[Can be found in /etc/fullpageos_version ALWAYS INCLUDE.]

#### Screenshot(s) showing the problem:

[If applicable. Always include if unsure or reporting UI issues.]

guysoft, 2017-09-20:
1. What user did you change?
2. What error do you get in ssh?

12nick12, 2017-09-20:
I changed pi and it just says authentication rejected.

12nick12, 2017-09-20:
I figured so. I just figured I'd start here. Thanks for the quick response

closed 2017-09-20 by 12nick12