guysoft/FullPageOS #168: Chromium doesn't display anymore after "apt update ; apt full-upgrade"

jens-maus, 2017-09-26:
#### What were you doing?

I installed a fresh 0.7.0 image of full page os and afterwards I logged in via SSH and then performed a "apt update" and afterwards "apt full-upgrade" to update the underlying Linux system. After reboot Chromium doesn't show any webpage anymore but a black screen only. However, the mouse cursor is properly displayed and via VNC the web page is displayed as well. Only on the HDMI monitor no webpage is displayed anymore?!?

#### What did you expect to happen?

I expected FullPageOS to behave normal.

#### What happened instead?

After reboot chromium starts with a black screen, but the mouse cursors are displayed correctly.

#### Was there an error message displayed? What did it say?

No error messages

#### Version of FullPageOS?

0.7.0

guysoft, 2017-09-26:
Assuming lighttpd broke and is not coming up, chromium only starts after it.
Can you paste the output of ``systemctl status lighttpd.service``.
Also, suggest you use the nightly if you want a new OS. Since there where changes to support Stretch AFAIK

jens-maus, 2017-09-26:
Chromium is definitively running, otherwise I wouldn't probably see the web browser content in the VNC connection. Only on the HDMI connection the chromium display is completely empty (black). Only the cursor is actually visible and changes its style with moving the mouse over some content (by watching it via the VNC connection). Thanks for the comment about the nightly build. Will test this next time.

biohazd, 2017-10-04:
I think i have this issue too.

```
pi@ndc1-rpi02:~ $ sudo systemctl status lighttpd.service
● lighttpd.service - Lighttpd Daemon
   Loaded: loaded (/lib/systemd/system/lighttpd.service; enabled)
   Active: active (running) since Wed 2017-10-04 13:35:06 UTC; 27s ago
  Process: 803 ExecStartPre=/usr/sbin/lighttpd -t -f /etc/lighttpd/lighttpd.conf (code=exited, status=0/SUCCESS)
 Main PID: 843 (lighttpd)
   CGroup: /system.slice/lighttpd.service
           ├─843 /usr/sbin/lighttpd -D -f /etc/lighttpd/lighttpd.conf
           ├─855 /usr/bin/php-cgi
           ├─856 /usr/bin/php-cgi
           ├─857 /usr/bin/php-cgi
           ├─858 /usr/bin/php-cgi
           └─859 /usr/bin/php-cgi

Oct 04 13:35:06 ndc1-rpi02.ops lighttpd[803]: Syntax OK
Oct 04 13:35:06 ndc1-rpi02.ops systemd[1]: Started Lighttpd Daemon.
```

`  818 ?        Ssl    0:00 /usr/sbin/lightdm
  836 tty7     Ss+    0:01  \_ /usr/lib/xorg/Xorg :0 -seat seat0 -auth /var/run/lightdm/root/:0 -nolisten tcp vt7 -novtswitch
  864 ?        Sl     0:00  \_ lightdm --session-child 12 15
  877 ?        Ss     0:00      \_ /bin/bash /home/pi/scripts/run_onepageos
  895 ?        Ss     0:00          \_ /usr/bin/ssh-agent /home/pi/scripts/run_onepageos
  900 ?        S      0:00          \_ matchbox-window-manager
  901 ?        S      0:00          \_ x11vnc -many -rfbauth /home/pi/.vnc/passwd
  907 ?        S      0:00          \_ /bin/bash /home/pi/scripts/start_chromium_browser
  909 ?        Sl     0:08              \_ /usr/lib/chromium-browser/chromium-browser --disable-quic --enable-fast-unload --enable-tcp
  928 ?        S      0:00                  \_ /usr/lib/chromium-browser/chromium-browser --type=zygote --noerrdialogs --ppapi-flash-p
  934 ?        S      0:00                  |   \_ /usr/lib/chromium-browser/chromium-browser --type=zygote --noerrdialogs --ppapi-fla
  968 ?        Sl     0:20                  |       \_ /usr/lib/chromium-browser/chromium-browser --type=renderer --disable-gpu-compos
  971 ?        Sl     0:03                  |       \_ /usr/lib/chromium-browser/chromium-browser --type=renderer --disable-gpu-compos
 1031 ?        Sl     0:00                  \_ /usr/lib/chromium-browser/chromium-browser --type=gpu-process --enable-features=Autofil
 1033 ?        S      0:00                      \_ /usr/lib/chromium-browser/chromium-browser --type=gpu-broker `

bortoelnino, 2017-10-09:
I had this yesterday. I could get it to run again by removing the --kiosk arg in the start_chromium_browser script.

eagleapex, 2017-10-12:
I had the same problem. I was able to see the browser when turning off kiosk mode. But now I can see the interface. How can I get a nice clean browser?
I want to display a Google Photos slideshow.

guysoft, 2017-10-12:
First guys - try not to update. You can always re-flash back before the update.

@eagleapex What do you mean "the interface"? You mean a login screen?

guysoft, 2017-10-12:
I think the issue is because lightdm.conf is overwriting the automatic login settings that is done here:
https://github.com/guysoft/CustomPiOS/blob/CustomPiOS/src/modules/gui/end_chroot_script

eagleapex, 2017-10-12:
Sorry, I should have been more clear. I couldn't see the page rendered with the --kiosk mode. When I VNCed into the RPI, I was able to see the mouse cursor rendered over a black screen. It even changed from pointer, to ibeam and hand based on the hover context. 
Now that I turned off --kiosk, I can see the "interface" which is the chrome UI including address bar and status bar.
I only updated to try and keep the device secure.

eagleapex, 2017-10-14:
I tried reflashing back to fresh 0.7.0 and now the RPi ZeroW won't display anything over HDMI after it's initial reboot. I had this almost setup this evening with everything good except for the visible address bar. Now I'm worse than when I started. 
I'm trying to make a digital picture frame with http://dakboard.com/ for my wife who is going back to work from maternity leave. I guess I give up on this project now. I dunno. I can do that hardware so well, but the software enver works with these projects.

guysoft, 2017-10-14:
Please use 0.8.0 nightly. Its more stable than 0.7.0

eagleapex, 2017-10-15:
0.8.0 was able to connect to my 7"hdmi display and show my dakboard page with no problem. Thanks!

guysoft, 2017-10-15:
Ok, im just gonna release 0.8.0 because its far better

closed 2017-10-15 by guysoft

eibei88, 2017-11-09:
Having the same issue since I did an apt-get update & upgrade. Chromium was working fine till then, now all I see is the black screen on my HDMI device. If I connect via vnc everything works and displays just fine. Removing the --kiosk argument as mentioned above didnt help either.

Did anyone come to a (remote) solution yet? I'm not very keen on flashing the Raspberry to solve this problem, there has got to be another way??

biohazd, 2017-11-10:
@eibei88  are you on version 0.8.0 ?

eibei88, 2017-11-10:
No, it's an older version but the RPI is running far away and flashing to a new version is not the fast solution I'm hoping for... are the problems definitely solved in version > 0.8.0?

biohazd, 2017-11-10:
@eibei88 0.8.0 fixed my issues with screens going blank and upgrades.

eibei88, 2017-11-14:
@biohazd I installed the 0.9.0 now and everything works as it should, hope it stays that way, thanks :)

biohazd, 2017-11-14:
excellent :)