guysoft/FullPageOS #169: Scrolling screen on large display

Imburr, 2017-09-28:
Crosspost: https://www.raspberrypi.org/forums/viewtopic.php?f=28&t=194132

I am running FullPageOS which is built off of Raspbian: https://github.com/guysoft/FullPageOS

When I run this at my house on monitors and TV,s it works flawlessly.

When I run this at my church, through an HDMI to SDI converter, it slowly scrolls upwards. And the scrolling happens from boot, and during boot. When I plug in my laptop or other devices into the converter, all works well. I do NOT think it is the converter, but rather the HDMI output freaking out at the size of the TV maybe? I don't know if they are 4k, but they are 55-75 inches.

When I say scroll, its like the whole frame scrolls. There is a black horizontal bar where the display ends which is normal under the bezel- this slowly moves upwards, and everything on the screen goes with it in an infinite loop, even the mouse.

Any ideas? I tried messing with all of the config.txt settings to no avail.

Running: 2017-06-21-fullpageos-jessie-lite-0.7.0 on a RPi 3 B+ latest edition

guysoft, 2017-09-28:
I never used an SDI converter, but it seems from google that they require a frame synchronizer. Also, they seem to have a bunch of buttons to tune the framerate, can you post the model or picture of the converter?
Moreover, I am not sure what setup you have there, but you might have a way to use the composite connector on the raspberrypi instead of HDMI. Composite is also analog and there might be an easier way to convert it to work with SDI with the equipment you have there.

Also, a picture would be cool of the setup you have there :)
Will note this is not really a FullPageOS issue because its a video hardware issue. But will try to help because getting my code in to a church seems pretty cool.

Imburr, 2017-09-28:
Here is the [SDI Converter by Black Magic](https://www.bhphotovideo.com/bnh/controller/home?A=details&O=&Q=&ap=y&c3api=1876%2C%7Bcreative%7D%2C%7Bkeyword%7D&gclid=CjwKCAjwu7LOBRBZEiwAQtfbGOZgLJSboWYN809m3fuucv5SoTl9m9CeB-ZP0iohibZpFtn3BQQ61hoCiicQAvD_BwE&is=REG&m=Y&sku=1044816). I do not believe there are any buttons on it. It plugs into an AV distribution stack, which then sends signal to like 20 television sets mounted on the wall. I can get you a picture of the stack a little later today.

guysoft, 2017-09-28:
Ok, seems like you need ti force an hdmi output resolution and format. You said you played with ``config.txt``. What combinations of ``hdmi_group`` and ``hdmi_mode`` have you tried?

Imburr, 2017-09-28:
All of the HDMI modes I tried were basically random, and I saw no difference in the display for any of them- same resolution, etc. I almost think that the mode options were not even enabled since it was not making a change.

I tried the following options:
hdmi_group=2
--
hdmi_drive=2
--
hdmi_mode=1
--
hdmi_mode=4
--
hdmi_mode=16
--
hdmi_mode=20
--
sdtv_mode=0
--
hdmi_safe=1

Here is the converter: http://cubeupload.com/im/ruKeo7.jpg
Here are the inputs and outputs: http://cubeupload.com/im/uY9eVk.jpg http://cubeupload.com/im/UJqRIX.jpg
The Pi: http://cubeupload.com/im/Qys3CH.jpg
The video stack: http://cubeupload.com/im/s2bqLD.jpg
The distribution unit: http://cubeupload.com/im/Dk3cxb.jpg

My church is very modern and tech oriented, and we do tons of stuff with media display. We are in the process of launching a second campus, so getting this Pi working would be super awesome... the new campus will be temporary for 12 months, so that means Sunday AM we will have a van filled with rolling racks, and will roll them into a school and deploy our whole campus in a couple of hours. Using point-to-point to stream live video and audio streams to the second campus. Being able to mount a Pi to a rolling rack to display our digital signage without having a whole complex setup would save us a LOT of time and money.

Side note, we do not have any sort of analog input anywhere, we are all digital. I could output analog from the pi to a converter and then into the SDI converter, but not sure if this would change anything.

The AV tech tried to upscale the HDMI from the Pi (which is 1080i) to 1080p and nothing changes. He also downscaled it to 720p with no change.

I suppose I could purchase [something like this](https://www.bhphotovideo.com/c/product/558906-REG/Blackmagic_Design_CONVMAAS_Mini_Converter_Analog_to.html) and output Analog from the Pi? I would much prefer not to purchase something like this, as it partially negates the benefit of using a Pi (cheap).

Imburr, 2017-09-28:
Also, here is a video of what the display is doing. I recorded it on the display of the AV distribution unit, but whatever is on this screen is what is also on all of the displays.

https://youtu.be/MCJnjD0DgaM

Imburr, 2017-09-28:
And here is the Xorg log: https://pastebin.com/f57nijpM

If I look at the supported display info modeset(0): Modeline "1920x1080i"x60.0   74.25  1920 2008 2052 2200  1080 1084 1094 1125 interlace +hsync +vsync (33.8 kHz eP) it should be running 1080i at 60Hz which is hdmi_mode=5.

guysoft, 2017-09-28:
Strange,  I was assuming SDI was aan analog signal and that the scrolling was from a wrong framerate (aka frame syncronizer). But it looks smooth. Did you try contacting Black Magic? They might br able to support their product.
Also any ideas @tailorvj ?

Imburr, 2017-09-28:
I can reach out to them.

Imburr, 2017-09-28:
Do you think it worthwhile to test with the Stretch image? That is what I originally put on the card, but I had no display show up at all so I rolled back to Jessie.

guysoft, 2017-09-28:
I think you should open an HDMI table and figure out what frequently to
use. Also try different resolutions, because if its a screen refresh issue
the scrolling speed should change.

And the current nightly is worth trying too, chromium 60 which is there is
much more stable

Imburr, 2017-09-28:
How do I do that? I thoguht reading the Xorg output which lists the supported resolutions was how I figured out which resolution the TV supports?

Strange thing is, no matter what I change the mode or group too, nothing changes on the display- at always looks the same.

Imburr, 2017-09-29:
We resolved this. Black Magic did not know what the issue was. We ordered a new SDI converter and while installing it, the AV tech noticed that the one we were using was 4K. After we swapped it out, everything works great.

SO apparently outputting from HDMI on the Pi into a 4K SDI converter is no bueno. So glad this is fixed, going to scale this out in a major way over the next 12 months!

guysoft, 2017-09-29:
Strange, the SDI spec said it auto-detects the HDMI resolution. Cool to know you worked it out. Closing.

closed 2017-09-29 by guysoft