guysoft/FullPageOS #170: update wording on page fro location of fullpageos-network.txt

biohazd, 2017-09-29:
Can you update the wording on this line of the main page
"Configure your WiFi by editing fullpageos-network.txt on the root of the flashed card when using it like a flash drive"

Change it to read **"/boot"** and not **"root"**

guysoft, 2017-09-29:
1. You are welcome to send a pull request
2. The wording is like that because windows users only see the boot
partition and thats why the stuff is all there

guysoft, 2018-02-16:
Closing if no reply

biohazd, 2018-02-17:
PR submitted.

guysoft, 2018-02-17:
@biohazd See point 2. The wording is like that because windows users only see the boot partition and thats why the stuff is all there. Changing it to `/boot` will confuse users.

biohazd, 2018-02-17:
ahhh, please reject my PR and leave as is :)

guysoft, 2018-04-11:
Fixed by https://github.com/guysoft/FullPageOS/commit/ff0d6c4cf3f46a85febd2edf07970afc96869347

closed 2018-04-11 by guysoft