guysoft/FullPageOS #171: Pulsing Resolution - HDMI to SDI

Imburr, 2017-10-10:
#### What were you doing?

I posted another thread about using an HDMI to SDI converter, which was solved by replacing the converter with a non-4K version. With the standard HD version however, the display pops up just fine, but it is shoddy resolution- like text is only partially legible. Every 3-5 seconds, it pulses, and the resolution sharpens for a few seconds, then reverts. I Have played with the group and forced resolution settings but nothing I do changes anything on the output. Anyone have any suggestions even if only on how to troubleshoot?

#### Version of FullPageOS?

0.7.0

#### Screenshot(s) showing the problem:

Can post video if needed.

guysoft, 2017-10-10:
Really hard to see. Looks like an analog signal, perhaps interpolation of the hdmi signal. Id ask black magic again.

guysoft, 2018-02-16:
Hey, was this solved in the end? I remember you said something about the company of the SDI fixing it.

Imburr, 2018-02-16:
Yep, it was caused by the BlackMagic HDMI to SDI converter being 4k. As soon as we moved to a normal one, worked great.

[The Dill Design]
Chris Dill | SEO Consultant
Office: (804) 419-7445
Mobile: (804) 647-0292

[Facebook]<https://www.facebook.com/TheDillDesignSEO/> [Google +] <https://plus.google.com/+Thedilldesign/about>  [LinkedIn] <http://www.linkedin.com/company/3284603>  [Twitter] <https://twitter.com/TheDillDesign>  [Yelp] <http://www.yelp.com/biz/the-dill-design-llc-richmond>  [YouTube] <https://www.youtube.com/channel/UCNl4aL4dpwNMclTbjVKxLVQ>
CONFIDENTIALITY NOTICE: This e-mail message and any attachments are intended solely for the use of the intended recipient, and may contain information that is confidential, privileged and exempt from disclosure under applicable law. If you are not the intended recipient, you are not permitted to read, disclose, reproduce, distribute, use or take any action in reliance upon this message and any attachments, and we request that you promptly notify the sender and immediately delete this message and any attachments as well as any copies thereof. Delivery of this message to an unintended recipient is not intended to waive any right or privilege.

closed 2018-02-16 by guysoft