guysoft/FullPageOS #173: WiFi connection broken in 8.0.0

schliflo, 2017-10-17:
#### What were you doing?

I'm using FullPageOS on a RasPi 3 to power my photobox display (via https://github.com/schliflo/flashair-photobox). Therefore the Pi has to connect to the FlasAir-card's WiFi to display the page. This was working perfectly fine in 7.0.0

#### What did you expect to happen?

Pi connecting to the specified WiFi

#### What happened instead?

No connection established

#### Was there an error message displayed? What did it say?

Chrome's "there is no connection to the internet connection" page with the dinosaur

#### Version of FullPageOS?

0.7.0 (i double checked this; this is wrong for [0.8.0](https://github.com/guysoft/FullPageOS/blob/0.8.0/src/modules/fullpageos/filesystem/root/etc/fullpageos_version) and [0.8.8](https://github.com/guysoft/FullPageOS/blob/0.8.8/src/modules/fullpageos/filesystem/root/etc/fullpageos_version))

pspEgg, 2017-10-26:
I too have the same exact issue. 

While booting an error message “failed to start secure ssl server” appears as well.

petrleocompel, 2017-11-29:
I had troubles to connect also. I tried to configure fullpageos-network.txt but then i ended up with `wpa-supplicatn` configuration. And it worked. Give it try.

guysoft, 2018-02-16:
There is a solution here and no reports if it works. Did anyone test this? Can I close the issue?

schliflo, 2018-02-17:
As configuration via `fullpageos-network.txt` - like stated in the docs - does not work I would this still consider to be a bug.
I myself downgraded to 0.7 where it does still work. I can test a newer build though if this should be fixed there.

guysoft, 2018-02-17:
did you try octopi-wpa-supplcant.txt?

guysoft, 2018-02-17:
Please use ``octopi-wpa-supplcant.txt`` its undocumented because I have no time, it will solve your problem. The file is self explanatory and I am keeping both methods because in some cases only one of the two work.

cc @foosel we should really put ``octopi-wpa-supplcant.txt`` as the default option,

marians, 2018-02-18:
I got wifi working after filling in my network details in both `fullpageos-network.txt` and `fullpageos-wpa-supplicant.txt`.

schliflo, 2018-02-19:
I can offer to check this in depth and update the documentation accordingly. But I'm currently really occupied with work so I cannot give an ETA... Will keep you updated.

ls009, 2018-02-21:
I had the same issues, `octopi-wpa-supplcant.txt` file with the same content as my `fullpageos-wpa-supplicant.txt` fixed the issue

guysoft, 2018-04-23:
Major changes in 0.9.0 nightly regarding firmware and kernel. Suggest you try now.

guysoft, 2018-07-21:
Closing if no reply

closed 2018-08-06 by guysoft