guysoft/FullPageOS #175: Browser credentials

numerounos, 2017-10-20:
Does FullPageOS support populating browser page credential population

guysoft, 2017-10-20:
Its chromium (currently version 60), it supports what chromium supports.

Atem18, 2019-08-09:
Did someone found a solution to that problem ?

guysoft, 2019-08-11:
I still don't understand what is the problem, OP didn't reply. Please describe what you are missing, what is excited, and what is the current result.

Atem18, 2019-08-12:
@guysoft Thanks for replying.

Well my use case is simple. I just want to display Grafana dashboards but to do so, you have to be loggued in. They don't support (yet) to generate a custom URL that allows you to view the dahsboard like you were loggued in.
The only thing that "works" is that if you pass a token via a header.

But I don't see how to do it without installing a chrome extension, which is not the goal of FullPageOS.

Do you have any advice ?

Thanks a lot in advance.

guysoft, 2019-08-12:
@Atem18 I have a grafana dashboard on my instance at work. Works fine.
I sometimes need to re-login. About once a month.

Atem18, 2019-08-12:
@guysoft Ok but then you need to login manually via VNC or so. My point was to do not have to login manually.

guysoft, 2019-08-12:
What about using something like this?
https://chrome.google.com/webstore/detail/auto-login/kjdgohfkopafhjmmlbojhaabfpndllgk?hl=en

Atem18, 2019-08-12:
Thanks, I thought about it but you still need interaction from the user. I wanted to be able to put the credentials somewhere via SSH, and that chrome would pick them up to login to a website.

guysoft, 2019-08-12:
You can use xdotool to do this in commandline. Look at the xdotool section in this guide:
https://pimylifeup.com/raspberry-pi-kiosk/

Atem18, 2019-08-12:
@guysoft Perfect, that's what I was searching for !

For your info and the next people, I modified the script "fullscreen" and added lines after xdotool mousemove 9000 9000, so the final script looks like this:

```bash
#!/bin/bash
export DISPLAY=:0
WID=$(xdotool search --onlyvisible --class chromium|head -1)
xdotool windowactivate ${WID}
xdotool key F11
xdotool mousemove 9000 9000
sleep 2
xdotool type 'username'
xdotool key Tab
xdotool type 'password'
xdotool key Return
sleep 5
xdotool click 1
```

The sleep 2 is to ensure that Grafana webpage is fully loaded, then we enter username and password and then the sleep 5 and click 1 is because chrome is asking to save password so we click anywhere to dismiss it.

Also as a final note, I modified run_onepageos to add
```bash
rm -rf /home/pi/.config/chromium
```
after the launch of VNC (so before the loop) to start fresh each time fullpageos boot.

Thanks a lot, you can close the issue. :)

guysoft, 2019-08-12:
Cool, closing also since OP isn't responding

closed 2019-08-12 by guysoft