guysoft/FullPageOS #177: On Boot I receive  Cannot login - Failed to start OpenBSD Secure Shell server

capqwerty, 2017-10-24:
#### What were you doing?

Booting the OS
#### What did you expect to happen?
System starts and I am able to ssh or VNC in

#### What happened instead?
Unable to SSH in and VNC client says password raspberry is incorrect
Also wireless does not work - not sure if it's related but wired does
not having a wireless issue on a very old FullPageOS

#### Was there an error message displayed? What did it say?
Failed to start OpenBSD Secure Shell server
#### Version of FullPageOS?

[Can be found in /etc/fullpageos_version ALWAYS INCLUDE.]
Unable to get to that folder as I cant escape chromium utilising the ALT F commands but I am utilising . .8.1 version

Hardware is a PI 3B

I would love to be able to get to a terminal session and get to the logs but can't seem to break free from chromium. Is there a way to stop the gui loading from config file in Boot?

guysoft, 2017-10-24:
Looks like its: https://github.com/guysoft/OctoPi/issues/424

What is the output of `ls -l /etc/ssh/ssh_host_*`?

capqwerty, 2017-10-24:
"What is the output of ls -l /etc/ssh/ssh_host_*?"  << Thats the problem I cant escape chromium to get to a terminal session

I would love to be able to get to a terminal session and get to the logs but can't seem to break free from chromium. Is there a way to stop the gui loading from config file in Boot?

guysoft, 2017-10-24:
Ctrl+alt+F1

To go back
Ctrl+alt+F7

capqwerty, 2017-10-25:
Thanks after changing the keyboard out from a Mac one to a pc one I can now
get to a prompt.
Username and password I assumed as pi/rasberry but it’s not accepting that

guysoft, 2017-10-25:
Password id wrong,  should be raspberry

capqwerty, 2017-10-25:
Ok confirming I can log in via SSH & VNC
I am getting in the boot log

FAILED] Failed to start OpenBSD Secure Shell server.
See 'systemctl status ssh.service' for details.

and running systemctl status ssh.service I get the following 

 ssh.service - OpenBSD Secure Shell server
   Loaded: loaded (/lib/systemd/system/ssh.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Wed 2017-10-25 00:03:51 UTC; 9h ago
  Process: 484 ExecStart=/usr/sbin/sshd -D $SSHD_OPTS (code=exited, status=255)
 Main PID: 484 (code=exited, status=255)

Oct 25 00:03:51 fullpageos systemd[1]: Starting OpenBSD Secure Shell server...
Oct 25 00:03:51 fullpageos systemd[1]: ssh.service: Main process exited, code=exited, status=255/n/a
Oct 25 00:03:51 fullpageos systemd[1]: Failed to start OpenBSD Secure Shell server.
Oct 25 00:03:51 fullpageos systemd[1]: ssh.service: Unit entered failed state.
Oct 25 00:03:51 fullpageos systemd[1]: ssh.service: Failed with result 'exit-code'.

and to answer your first question

pi@fullpageos:~ $ ls -l /etc/ssh/ssh_host_*
-rw------- 1 root root  668 Sep  8 01:54 /etc/ssh/ssh_host_dsa_key
-rw-r--r-- 1 root root  605 Sep  8 01:54 /etc/ssh/ssh_host_dsa_key.pub
-rw------- 1 root root  227 Sep  8 01:54 /etc/ssh/ssh_host_ecdsa_key
-rw-r--r-- 1 root root  177 Sep  8 01:54 /etc/ssh/ssh_host_ecdsa_key.pub
-rw------- 1 root root  411 Sep  8 01:54 /etc/ssh/ssh_host_ed25519_key
-rw-r--r-- 1 root root   97 Sep  8 01:54 /etc/ssh/ssh_host_ed25519_key.pub
-rw------- 1 root root 1675 Sep  8 01:54 /etc/ssh/ssh_host_rsa_key
-rw-r--r-- 1 root root  397 Sep  8 01:54 /etc/ssh/ssh_host_rsa_key.pub

guysoft, 2017-10-25:
Wait - so now you can ssh but the service claims ssh is not running m

capqwerty, 2017-10-25:
Correct

guysoft, 2017-10-25:
Not really sure how that can be. I mean how wouwyou ever know ssh if off if it rujs beyond the daemon.

Are you sure your not confusing ssh with login with a keyboard connected to the pi?

[image]

guysoft, 2017-10-29:
Fix found, building and will upload a fix to nightly

closed 2017-10-29 by guysoft