guysoft/FullPageOS #179: Sluggish CSS animations

tdusmoha, 2017-10-27:
#### What were you doing?

Loading a full page image slider

#### What did you expect to happen?

The images would slide smoothly

#### What happened instead?

Instead it was sluggish

#### Was there an error message displayed? What did it say?

No error, but the same page works fine on other distro.

#### Version of FullPageOS? 0.8.1

[Can be found in /etc/fullpageos_version ALWAYS INCLUDE.]

#### Screenshot(s) showing the problem:

[If applicable. Always include if unsure or reporting UI issues.]

guysoft, 2017-10-27:
The GPU acceleration is enabled by default in FullPageOS. Its the main
difference from other distros. You can try disabling it and see if it
improves the situation.

remi-grumeau, 2017-10-28:
Is the transition done using CSS transform? Did you hide the backface-visibility? Goofing with opacity?

If done by some crappy Swiss Knife JQuery plugin, remember a Pi as a 4x slower CPU & half RAM a mid-level smartphone has... you have do be smart regardless performance within your app.

Remi

tdusmoha, 2017-11-02:
@guysoft tried to tweak the settings on chrome://flags, but no success.
@remi-grumeau yes i'm using CSS transform.

For lack of a better solution, I am starting chromium over Jessie (desktop) in kiosk mode. My css animations work flawlessly.

If you would like to give it a go, here's the page i'm trying to load: redline dot mycloud dot mu

ppejovic, 2018-07-15:
@guysoft how do you go about disabling the gpu acceleration? Thanks.

tdusmoha, 2018-07-16:
Type the following in your address bar `chrome://gpu`

ppejovic, 2018-07-16:
@tdusmoha no, this is a distro setting.

@guysoft Is this the correct setting `GUI_INCLUDE_ACCELERATION=no`? That I assume would require me to build the distribution?

guysoft, 2018-07-16:
@ppejovic Indeed, this is the default setting when building the distro.

ppejovic, 2018-07-17:
Thanks @guysoft. I did a non-accelerated build and I saw a noticeable improvement in the css animations on my Pi 2 Model B. There is still some jitter, so will have to keep looking at what I can tweak next!

Amazing project btw. Thought it was cool building a distro on the os itself. Thank-you!

ppejovic, 2018-07-17:
So on the non-accelerated distro I enabled the "OpenGL desktop driver with full KMS" and now the css animations are buttery smooth! I haven't tested this setting on the accelerated version. HTH.

guysoft, 2018-07-22:
Hey, I created a variant of FullPageOS that has graphical acceleration disabled, so you can test that:

http://unofficialpi.org/Distros/FullPageOS/no-acceleration/nightly/2018-07-22_2018-06-27-fullpageos-no-acceleration-stretch-lite-0.10.0.zip

md5: [2197ebd78f65c17e4155e5bf93e7af55](http://unofficialpi.org/Distros/FullPageOS/no-acceleration/nightly/2018-07-22_2018-06-27-fullpageos-no-acceleration-stretch-lite-0.10.0.zip.md5)

tdusmoha, 2018-07-23:
Great! Thanks for sharing. Will test & post results

samijandali, 2018-07-25:
@guysoft thanks for referring me here!
Well for me now the page is crashing and I'm getting the "Aw, snap" from chromium (used the above image and enabled OpenGL with full KMS

samijandali, 2018-07-27:
@ppejovic I have my videos in an HTML video element and they run off using javascript. I am using Pi 3 B (not the latest one)

tvandergeer, 2018-08-24:
@ppejovic Can you elaborate on what I should do to enable the "OpenGL desktop driver with full KMS" option?

ppejovic, 2018-08-24:
@tvandergeer http://www.raspberryconnect.com/gamessoftware/item/314-trying_out_opengl_on_raspberry_pi_3

Though subsequent to my post above I’ve found this makes chrome crash during video playback.