guysoft/FullPageOS #180: Wifi Not working with Last release

conrad9900, 2017-11-01:
#### What were you doing?
Updating current images with the newest, downloaded the newest and planned to just copy over the txt files from my nearly 20 working pi's

[Please be as specific as possible here]

#### What did you expect to happen?
once the txt files were copied, it should boot as normal to wifi
#### What happened instead?
Boots to wired DHCP if i connect it, will not connect to wifi, network txt file contents is copied from an already working Pi

#### Was there an error message displayed? What did it say?
There is no internet Connection

#### Version of FullPageOS?
Latest

[Can be found in /etc/fullpageos_version ALWAYS INCLUDE.]

#### Screenshot(s) showing the problem:

[If applicable. Always include if unsure or reporting UI issues.]

Wifi Changes:

## WPA/WPA2 secured
    iface wlan0-fullpageos inet manual
    wpa-ssid "Peter_Net"
    wpa-psk "Profile1"

DNS & Domain:

domain profile.local
dns-nameservers 192.168.0.10

guysoft, 2017-11-01:
1. What version?
2. Please use the newer ``fullpageos-wpa-supplicant.txt``
3. revert the changes you made ``fullppageos-network.txt``

conrad9900, 2017-11-01:
Appreciate all the help mate, well done on an excellent piece of kit,

conrad9900, 2017-11-01:
2 other questions if you may: 1. What is the easiest way to flip the screen 180 degrees
2. What is the best way to reconfig and save the image with the wifi and image changes i need etc. I tried to clone an SD card that had all the config but it creates an image the full size of the SD card. (16GB)

guysoft, 2017-11-01:
Please do not mix questions. This is an issue tracker, not a support forum
Also, you are no answering my questions - I am not supporting unknown versions.

conrad9900, 2017-11-01:
The version i am using was downloaded yesterday.

guysoft, 2017-11-01:
There is a stable and nightly. When it was downloaded does not help. You ignored the instructions you posted:
``[Can be found in /etc/fullpageos_version ALWAYS INCLUDE.]``

conrad9900, 2017-11-01:
when i SSH to my pi, and navigate to my /etc, there is no file called fullpageos_version. I downloaded the stable version yesterday

facetime88, 2017-11-06:
Same here, wifi not working.

Gun-neR, 2017-11-08:
Finally got a visible screen... Now no WiFi 

**What version? -** 2017-10-22_2017-09-07-fullpageos-stretch-lite-0.8.1
**Please use the newer fullpageos-wpa-supplicant.txt -** Done, although it is not clear exactly which # to remove and which to keep, but I think I got the relevant ones.
**revert the changes you made fullppageos-network.txt -** Done