guysoft/FullPageOS #183: Fatal Error when restarting Chromium and not Pi

Imburr, 2017-11-10:
So, I have been restarting my Pi every 4 hours. I run a BlueIris camera server at my house with 3 cameras. The problem is that after some time, the page stops responding or prompts for creds and freezes. I have creds embedded in the URL. For the most part it works well.

What I would like to do is reboot Chromium once per hour.

I set this in my chrontab as root:

0 */1 * * * kill -9 $(ps aux | grep "start_chromium_browser" | grep -v grep | awk '{print $2}')

When I run it manually, all is good. When I leave the room for a few housrs and come back, its a black screen, or non-active monitor. Rebooting fixes all issues.

Expectation:  I expect Chromium to close, open, and go back to my URL.

Result: Blank screen.

#### Version: 0.7.0

#### Screenshot: https://file.thedilldesign.com/f/9157b5de9a82481eb001/

petrleocompel, 2017-11-29:
Why you wouldn't just trigger command in scripts for refresh page ? 
`/home/pi/scripts/refresh` It will refresh webpage. You dont have to kill whole start script.

guysoft, 2017-11-29:
Please use 0.8.1 which is much more stable.

closed 2017-11-29 by guysoft